Klipper3d/klipper #1401: [Question] Core XYU possible?

Ikes72000, 2019-03-16:
Hi,

I am currently designing a core XY, which will work under Klipper. I would like to add the possibility of having a core XYU on this printer, for future upgrades.

Is it possible to drive a core XYU with Klipper?

KevinOConnor, 2019-03-20:
I'm not familiar with the details of XYU, but it should be possible to add additional kinematics and/or additional extruders.  It does require some additional development work though - see docs/Code_Overview.md for the details.

-Kevin

Ikes72000, 2019-03-23:
Ok, I'll watch this.
For now I'm trying to install Klipper on a STM32F103 board to manage additional fans, but it's not easy.

Thanks Kevin. 

Ikes

closed 2019-03-23 by Ikes72000

khessels, 2020-10-20:
To bad this is inactive, started building a corexyu printer, and would love to do this with klipper. Now i'm forced to go rrf, don't get me wrong, its great firmware as well .. just, its not something i run on a pi is it...