Klipper3d/klipper #1402: Start / End Code

noxx2015, 2019-03-16:
hello,

at the moment i use my marlin start and end code. but i ve problem. 
#1 double home before printing
#2 problem after printing. at first the printer goes "home" and after this, the printer (nozzle) run in my printing part.

have anyone a start and end script or can check my script?

start script:
https://pastebin.com/raw/TAPziJ6D

end-script:
https://pastebin.com/raw/CrYnSYU6

printer.cfg
https://pastebin.com/raw/aygRewbm

thx

AxMod3DPrint, 2019-03-16:
If you want to home X&Y Separately from Z, you need to home them individually IE: 
```
G28X
G28Y
G28Z
```

G28XY or G28 X0 Y0 will not work.

closed 2019-03-24 by klipper-gitissuebot

jakep82, 2019-03-25:
@AxMod3DPrint What you've posted here will actually cause this printer to home every axis 3 times during the start script, and twice during his end script.  This issue is the homing override.

@noxx2015 The default setting for a homing override is to run it for all axes.  That means running `G28 X0` will actually home all 3 axes.  The solution to your problem is to simply use a single `G28` in your start script, and use `G1 X0 Y0` in your end script.

For more info read about the homing override at [this link.](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L1144)