Klipper3d/klipper #1404: "Must home axis first" during print

Montbel, 2019-03-17:
Hi,
I have a "Must home axis first" during print, in fact the last G1 of the print, but could not identify why, it seems to me that the command sequence is the same as other commands previously sent.
Attached Klippy.log, octoprint.log
printer.cfg and test.gcode in Archive.zip

Thanks,
Christian

[klippy.log](https://github.com/KevinOConnor/klipper/files/2974908/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/2974909/octoprint.log)

[Archive.zip](https://github.com/KevinOConnor/klipper/files/2974912/Archive.zip)

joonadec, 2019-03-17:
Looked through your klippy.log, the error message is - 

`Must home axis first: 98.149 103.799 1.000 [12.000]`

And I looked through the gcode, and the corresponding lines are here -

```
G0 F2700 X98.299 Y103.649
M204 S300
M205 X3 Y3
G1 F1290 X98.149 Y103.799 E21.10596
```

As you can see, its trying to extrude 20mm which not moving the nozzle by much. I would get this is what buggers up klipper.

Have you tried another slicer?

Hywelmartin, 2019-03-18:
In your end gcode you send M84 and then try to move ....
```
[gcode_macro END_PRINT]
gcode = 
	M117 End Printing.
	M104 S0
	M140 S0
	M84 ; disable motors<----------
	M107 ; turn fan off
	G92 Z0 ; zero Z
	G1 E-1 F300 ;retract the filament a bit before lifting the nozzle to release some of the pressure
	G1 Z0.5 E-3 ;X-20 Y-20 F1000 ;move Z up a bit and retract filament even more
	G1 X0 F2000 ;move X to min endstops so the head is out of the way
	G1 Y180 F2000
	M84 ;steppers off
	G90
	M300 P300 S4000
```

Montbel, 2019-03-19:
Thanks for your answers. 
@joonadec, the problem does not comes from this line, the extrusion is only a small fraction of mm.
@Hywelmartin , you got it !
I was focussed on the coordinates of the error and though it was the target coordinates, instead the error message is the source coordinates.
-> close the issue.

closed 2019-03-19 by Montbel