Klipper3d/klipper #1407: SET_TMC_CURRENT

workheart, 2019-03-17:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2975524/klippy.log)

I am looking to life change the current of my extruders, but it just gives back 
```
Send: SET_TMC_CURRENT STEPPER=extruder0 CURRENT=300
Recv: // Unknown command:"SET_TMC_CURRENT"
```
it is the same behavior for all steppers
DUMP_TMC STEPPER works okay,

jakep82, 2019-03-18:
Driver current is not run time configurable. I'm not sure where you found that extended gcode, but it's not valid.

amblidex, 2019-03-18:
That command is valid but only for the TMC2660. The g-code docs are confusing because the sentence `The following commands are additionally available when the "tmc2660" config section is enabled:` has a formatting problem and is part of the previous bullet instead of starting a new section for tmc2660 only commands

workheart, 2019-03-18:
Oka thats sad, i used to use it for my fillament changing code, but thanks for the fast reply.

closed 2019-03-18 by workheart