Klipper3d/klipper #1408: Enhancement Request: Specify amount of Cycles for PID Autotune

drphil3d, 2019-03-17:
I'm having a hard time tuning the PID of my 24v 90 watt hotend and noticed that the cycle time can't be specified. It would be helpful to be able to specify this from gcode. I'm also curious what is the procedure for pid tuning when you don't have baseline for the hardware you are using?

[image]

joonadec, 2019-03-17:
@drphil3d Have you tried setting a larger TUNE_PID_DELTA in pid_calibrate.py? 

https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/pid_calibrate.py

drphil3d, 2019-03-17:
I have not tried that, I understand what PID does but I'm not familiar with how the formula works.

drphil3d, 2019-03-17:
[klippy 90watt 24v hotend.log](https://github.com/KevinOConnor/klipper/files/2975873/klippy.90watt.24v.hotend.log)

drphil3d, 2019-03-18:
Ugh my hotend temp is swaying around by about 4 degrees according to my thermocouple

Muhata10111b, 2019-03-19:
If you still have the issue try using "smooth_time: 0.5" under the extruder section. If that doesn't work increase it to 4, default is 2. Had a similar issue and lowering it to .5 made the temperature more erratic, but I stopped getting the temperature swings.

jakep82, 2019-03-19:
90 watts is extremely powerful for a hotend.  I would use `max_power: 0.5` to limit it.  I would expect that to eliminate the temperature swings.

KevinOConnor, 2019-03-21:
I'm not sure what is being requested.  The pwm_cycle_time is available and documented (but it is not what you want).  If you're seeing oscillations in the temperature, that usually is the result of a poor pid tune.  As @jakep82 reports, a very powerful heater may make it difficult to get a good pid tune and reducing the max_power may help.

-Kevin

drphil3d, 2019-03-21:
I think the problem was my power supply. I noticed the led on it was
dimming at the same rate of the pwm cycle. 4 amps at 24v maybe it’s just
worn out.

KevinOConnor, 2019-03-31:
It looks like this issue was resolved, so I'm going to close this ticket.

-Kevin

closed 2019-04-01 by KevinOConnor