Klipper3d/klipper #1410: BMG presure advance and torque

wizhippo, 2019-03-19:
I have recently upgraded to an BMG extruder and am tuning pressure advance.

Has anyone else here done so and bee successful?

I have found that it maxes out at 2000mm/s acceleration where as my stock extruder had no issue with 4000mm/s.

I' guessing this is due to the BMG requiring more torque?   I just want to confirm that my thoughts are correct here.

theopensourcerer, 2019-03-19:
The BMG is 3:1 geared, so it doesn't need more torque; it requires less. But the trade-off is that the motor must run 3 times as fast to extrude the same amount of filament. I'm using a BMG clone in direct, not bowden, and have a PA of 0.07 for PLA and 0.2 for PETG and it works fine. Not sure what my accel is set to though as am not at home.

wizhippo, 2019-03-19:
Thought I may have had it backwards. Guess my motor/driver just can’t handle the 3x speed increase needed.  Odd though as retraction has no issue. Guess that’s because PA needs very quick moves compared to retraction. 

I too can do PA below .2 but unfortunately I need .4 due to my Bowden.  Guess I’ll just have to not use PA for now or print really slow. 

Thanks you for chiming in.

theopensourcerer, 2019-03-19:
It could be something as simple as your e-steps and/or micro-steps. Attach your klippy.log and then we can all take a look and possibly offer further advice.

theopensourcerer, 2019-03-19:
@wizhippo I just checked my printer.cfg:
```
[printer]
kinematics: corexy
max_velocity: 250
max_accel: 1500
max_z_velocity: 10
max_z_accel: 100
```
I use TMC2208 drivers on X, Y and Z and a DRV8825 on E set for 1/32 uSteps, with a calibrated step distance of 0.001180.

HTH

wizhippo, 2019-03-19:
@theopensourcerer Looks very similar to my PA working settings.

Going from 4000mm/s to 1500mm/s acceleration is a bit of a trade off I guess.  I'll do a few test prints and see how significantly it effects print time and decide if to continue with PA or not.

closed 2019-03-19 by wizhippo

banolka, 2019-05-03:
What are you stepper distance settings for bmg?

wizhippo, 2019-05-03:
> What are you stepper distance settings for bmg?

[extruder]
step_distance: 0.0024145313708312

tmc2130 extruder]
microsteps: 16