Klipper3d/klipper #1411: Top level menu won't stay displayed

w1ebr, 2019-03-19:
The top level menu that shows extruder temps won't stay displayed on the screen. It jumps to the next level of menu. When I select"main menu", it displays for about 1 second and then goes back to the next level of menu.

Has anyone seen this behavior?

[klippy.log](https://github.com/KevinOConnor/klipper/files/2981168/klippy.log)

closed 2019-03-22 by w1ebr

mbgroot, 2019-04-24:
I have the same problem. Did it work for you?

w1ebr, 2019-04-24:
This behavior was because I didn't have the signal inversion state set up properly.
This is what I have now:
[display]
lcd_type = st7920
cs_pin = PD2
sclk_pin = PD3
sid_pin = PL7
encoder_pins = ^PB5, ^PB6
click_pin = ^!PL6

w1ebr, 2019-04-30:
Yes, look at my post, I updated it with my solution. Gene

mbgroot, 2019-04-30:
Thank you! Thank you! I also solved the problem, but forgot to write :)