Klipper3d/klipper #1412: Internal error on command:"QUERY_ENDSTOPS"

inventowiz, 2019-03-19:
I recently updated to master and was unable to issue QUERY_ENDSTOPS nor QUERY_PROBE. Seems to be an issue with XOR on types.

```Internal error on command:"QUERY_ENDSTOPS"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 195, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 75, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/query_endstops.py", line 26, in cmd_QUERY_ENDSTOPS
    out.append((name, mcu_endstop.query_endstop_wait()))
  File "/home/pi/klipper/klippy/mcu.py", line 241, in query_endstop_wait
    return self._last_state.get('pin', self._invert) ^ self._invert
TypeError: unsupported operand type(s) for ^: 'str' and 'int'
```
After digging a bit more into this, I discovered that `self._last_state.get('pin', self._invert)` is returning `PA0`, which corresponds to `ar22` on the arduino mega board. I am using a RAMPS 1.4, and this pin is disconnected in hardware. 

Could this be an issue with my configuration?

See log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2981249/klippy.log)

cahayez, 2019-03-19:
I have this issue as well

cahayez, 2019-03-19:
Internal error on command:"QUERY_ENDSTOPS"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 195, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 75, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/query_endstops.py", line 26, in cmd_QUERY_ENDSTOPS
    out.append((name, mcu_endstop.query_endstop_wait()))
  File "/home/pi/klipper/klippy/mcu.py", line 241, in query_endstop_wait
    return self._last_state.get('pin', self._invert) ^ self._invert
TypeError: unsupported operand type(s) for ^: 'str' and 'int'

[klippy.log](https://github.com/KevinOConnor/klipper/files/2983380/klippy.log)

KevinOConnor, 2019-03-19:
Thanks.  Should be fixed now (commit 746a378c).

-Kevin

inventowiz, 2019-03-19:
I can confirm that this issue is fixed in master. Thanks Kevin!

Closing.

closed 2019-03-19 by inventowiz

cahayez, 2019-03-20:
also confirmed. Thanks!