Klipper3d/klipper #1415: After running bed mesh calibration, homing sequence inverts Z axis direction

cassianoleal, 2019-03-19:
After running a bed mesh calibration, I ran a `G28` to home all axis.

At that point, X and Y homed fine but the Z axis started going up instead of down.

I had to run to the printer and cut the power otherwise the gantry would ram into the top of the printer.

This started happening after having updated Klipper to commit 06c0463d449f0cb4d90b3c72eff19d8950d0eefd from 378402b4d68c41fb74c02164d10b9161f65a1b4d, updating the firmware and restarting `klipper`.

I'm running Klipper on a MKS Gen L board with all 4 drivers TMC2208's in UART mode.

Bed mesh calibration configuration:

```
M104 S130
M140 S60
M190 S60
M109 S130

G91
G28 Z0
G90
BED_MESH_CALIBRATE
@BEDLEVELVISUALIZER
BED_MESH_OUTPUT
BED_MESH_PROFILE SAVE=octoprint
SAVE_CONFIG
```

Find attached `klippy.log` from both the bad config and the good one:

[bad_klippy.log](https://github.com/KevinOConnor/klipper/files/2985206/klippy.log)

[good_klippy.log](https://github.com/KevinOConnor/klipper/files/2985267/good_klippy.log)

KevinOConnor, 2019-03-20:
This was a serious error that was fixed in commit d7665875.

-Kevin

cassianoleal, 2019-03-20:
Wow, thanks for the quick response! I'll test it tonight and feed back here.

cassianoleal, 2019-03-22:
Confirmed working, thanks!

closed 2019-03-22 by cassianoleal