Klipper3d/klipper #1417: "restart" stepper movement wrong

fhtagnn, 2019-03-19:
Tested repeatedly on a Voron with the latest master freshly pulled, fresh install per installation readme, both ramps freshly flashed. After service restart / boot (best tested with slow homing speeds): 

1. G28: printer homes as expected
2. Move to safe location (in may case x175 y175 z 150)
3. Issue a "restart" (say config changes)
3. G28: y moves forward instead of x right ultimately crashing the toolhead into the gantry front idlers

Initially, my homing override started with a z move up. Stress tested my gantry due to front steppers moving up and back steppers moving down.

FHeilmann, 2019-03-19:
I also encountered this issue.

shaiss, 2019-03-20:
I have the same issue, but only after I did a git pull to the latest version today. my config @ https://pastebin.com/uyCnygFZ

KevinOConnor, 2019-03-20:
Thanks.  That's a bad error.  Fix is in commit d7665875.

-Kevin

markwinger, 2019-03-20:
I am new to klipper(1 day).  I am seeing this problem.  How do I get this fix installed?

theopensourcerer, 2019-03-20:
It's in the docs but basically something like this (from memory so please check):

cd klipper
sudo service klipper stop
git pull (If you are on the master branch - use git status to check)
make clean
make
make flash FLASH_DEVICE=[your-usb-device-name]
sudo service klipper start

fhtagnn, 2019-03-20:
Tested the fix. All back to normal!

Thanks, Kevin

closed 2019-03-20 by fhtagnn

markwinger, 2019-03-22:
Worked for me too.