Klipper3d/klipper #1423: error: Move exceeds maximum extrusion (1.100mm^2 vs 0.640mm^2)

bumbalu, 2019-03-20:
Hello, Kevin. 
First, thank you for nice software and constant efforts to maintain it!

The error happens on the same print, at the same place. It is a tube standing vertically and the error happens exactly when the nozzle should lift up to the second layer. I tried to move the model on the table, re-slice it - no effect. 

G-code from terminal, just prior to error: 
=====
Send: N463 G1 X121.96 Y90.019 E40.58531 F134.6*25
Recv: ok
Recv: ok
Send: N464 G1 X120.085 Y90.46 E40.58531 F4200*63
Send: N465 G1 X120.145 Y90.462 E40.68812 F131.7*34
Recv: ok
Send: N466 G1 X120.325 Y90.468 E40.79622 F383.7*41
Recv: ok
Send: N467 G1 X120.624 Y90.479 E40.90722 F611.3*32
Recv: // Move exceeds maximum extrusion (1.444mm^2 vs 0.640mm^2)
Recv: // See the 'max_extrude_cross_section' config option for details
Recv: !! Move exceeds maximum extrusion (1.444mm^2 vs 0.640mm^2)
Changing monitoring state from "Printing" to "Cancelling"
Send: N468 M108*32
Send: N469 M84*20
Recv: ok
Send: N470 M104 T0 S0*34
Recv: // Move exceeds maximum extrusion (1.100mm^2 vs 0.640mm^2)
Recv: // See the 'max_extrude_cross_section' config option for details
Recv: !! Move exceeds maximum extrusion (1.100mm^2 vs 0.640mm^2)
=====

I am attaching the klippy.log, error is on the line 15894
[klippy.log](https://github.com/KevinOConnor/klipper/files/2988824/klippy.log)

Anything else I can do on my side to help finding the cause?
Thank you again.

bumbalu, 2019-03-20:
Sorry, I didn't want to make font that big, is somehow happened by itself. 
Just to add: the problem always happens at layer change. I got rid of seam, but it persists. I tried to change the preloadVE settings (in KISSlicer) because extrusion/retraction happen at layer change. Now the error appears not at first but at third layer change.

megablue, 2019-03-20:
you shouldn't use kisslicer's preloadVE on Klipper. preloadVE try to pressurized the nozzle by extruding a lot during the start of a print. klipper doesn't like a slicer manipulating the extrusion like that. the best bet is just diasble preloadVE and use Klipper's pressure advance.

I should say, you shouldn't use kisslicer on klipper at all. the reason being kisslicer will generate code that slowdown the print speed during overhang and klipper doesn't like speed transition at all, it will result in a micro pause during the transition and again when it transits from slow to fast.

bumbalu, 2019-03-20:
After playing with different parameters I've found a relation between error and preloadVE parameter from KISSlicer. Bigger is the preloadVE earlier happens the error. If preloadVE is set to zero, the error seem not happen. But I wonder if Pressure Advance alone would be enough to prevent underextrusion right after the seam.

bumbalu, 2019-03-20:
megablue, thank you from bumbalu! :-) World is not perfect. Either Klipper or KISSlicer. I'm not sure what will I keep. 
Should we consider the topic closed or will Kevin comment something on this?

bruce356, 2019-03-21:
As recommended by @megablue do not use kisslicer with klipper, it will just disappoint you.
Use another slicer like S3D.
regards bruce

KevinOConnor, 2019-03-21:
As mentioned, preloadVE is not compatible with Klipper.  If you have additional information on how to disable that in KISSlicer, I can update https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md to advise future users.

-Kevin

bumbalu, 2019-03-21:
Kevin, thank you for pointing this link. I have missed it somehow.

closed 2019-03-21 by bumbalu