Klipper3d/klipper #1424: Lost communication with MCU 'mcu'

JustAnother1, 2019-03-20:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2989015/klippy.log.zip)

The print went on for 13 Hours without an issue and then it suddenly stopped.

I did several successful prints on the machine. The USB cable is of good quality and short. Octoprint also does not complain on the power supply. I therefore assume that I did everything I could.

I can not decode the log file. Can anybody tell me what the issue was?

hevilp, 2019-03-20:
I had it too, you are running a raspi? Maybe change sd card, I did it just this hours because of the same problem

JustAnother1, 2019-03-20:
@hevilp Yes this runs on a raspberryPi. I don't want to change the SD Card just on a feeling. Did you check the log file. Was there some indication that the SD card could be the issue?

hevilp, 2019-03-20:
I understand you, I changed my sdcard and it still happens! I changed usb cable, raspi, sd card, power for rpi, and still got the problem, I need help here too

hevilp, 2019-03-20:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2989362/klippy.zip)

hevilp, 2019-03-20:
I just swapped the usb port from mcu:mcu, so far it it running flawless

KevinOConnor, 2019-03-21:
There's some info on this in the FAQ: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors

These errors occur at the USB hardware and/or operating system level.  Beyond the info in the FAQ, we don't have any additional information on the root cause.

-Kevin

JustAnother1, 2019-03-22:
I have now used zip ties to fix the USB cable to the raspberryPi. So next time this occurs I can rule out all of the things mentioned in the FAQ.

KevinOConnor, 2019-04-07:
Okay, I'm going to close this issue as it looks like the discussion has ended.

-Kevin

closed 2019-04-07 by KevinOConnor