Klipper3d/klipper #1425: Question about Pressure Advance and small features

wizhippo, 2019-03-20:
Few questions:

Are `max_extrude_only_velocity`, `max_extrude_only_accel` respected by PA? I'm guessing not as it would need to keep up to the print acceleration and speed.

I have PA working to an extent but parts with small features such as a 2mm x 2mm column that tapers to a point cause me issues and I believe it is because my stepper (3:1 BMG extrude driving bowden) cannot keep up with so many quick changes in a short period of time.  Is this where `pressure_advance_lookahead_time` would benefit by increasing?

jakep82, 2019-03-20:
As the name implies, they only apply to extrude only moves which would be retraction and de-retraction.  Any move that involves PA will also involve X and/or Y movement.

wizhippo, 2019-03-20:
Answered thank you.

closed 2019-03-20 by wizhippo