Klipper3d/klipper #1429: Request: command to initialize tmc2208 steppers

wormiedk, 2019-03-21:
I have the 2208 drivers which get configured via UART. I have a power supply which gets turned on when starting a print. 
It would be great if I could add a command in my startup code to initialize the drivers. As it is now I need to issue a restart command to initialize the drivers

FHeilmann, 2019-03-21:
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#tmc2130-tmc2660-and-tmc2208

Use the INIT_TMC gcode

wormiedk, 2019-03-22:
That must have been added recently 👍 . I will update my firmware. Precently I get "UNKNOWN COMMAND" when I try to INIT_TMC :)

wormiedk, 2019-03-22:
Just tried updating to newest by using git pull and compiliing and uploading. I still get:
Send: TMC_INIT stepper=stepper_x
Recv: // Unknown command:"TMC_INIT"
Recv: ok

Although TMC_DUMP works fine:
Send: DUMP_TMC stepper=stepper_xRecv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 00081309 IHOLD=9 IRUN=19 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000000Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20

AxMod3DPrint, 2019-03-22:
> Just tried updating to newest by using git pull and compiliing and uploading. I still get:
> Send: TMC_INIT stepper=stepper_x
> Recv: // Unknown command:"TMC_INIT"
> Recv: ok

The command is `INIT_TMC STEPPER=<name>` You've used TMC_INIT ;)

wormiedk, 2019-03-22:
Ah stupid mistake :) Never mind me!

closed 2019-03-22 by wormiedk