Klipper3d/klipper #1430: Move out of range after pause resume

Twuen, 2019-03-21:
Hello,

I tried to insert a Gcode Script in Octoprint after initiating Pause and before resuming to lift up the nozzle and retract relative to the current printing position in order to not melt the print while pausing: 
G1 F3000 Z18
G1 F3000 Z-18
The move back before resume of the z-axis always triggers a error "Move out of range".

I just updated to 0.7 before I never had this issue. Any Ideas? 

Thanks in advance.

Send: N85 G1 X297.518 Y298.871 E53.03987*97
Recv: ok
Send: N86 G1 X292.243 Y303.036 E54.69745*109
Recv: ok
Send: N87 G1 X292.087 Y303.156 E54.74599*110
Recv: ok
Send: N88 G1 X284.957 Y308.478 E56.94026*109
Changing monitoring state from "Printing" to "Pausing"
Recv: ok
Send: N89 G1 X284.324 Y308.901 E57.12802*96
Recv: ok
Send: N90 M400*30
Recv: ok
Send: N91 M114*31
Recv: X:284.324 Y:308.901 Z:0.200 E:57.128
Recv: ok
Send: N92 G90*43
Recv: ok
Send: N93 G1 F3000 Z18*4
Recv: ok
Send: N94 G91*44
Recv: ok
Changing monitoring state from "Pausing" to "Paused"
Send: M105
Recv: ok T0:209.7 /210.0
Send: M105
Recv: ok T0:209.1 /210.0
Changing monitoring state from "Paused" to "Resuming"
Send: N95 G90*44
Recv: ok
Send: N96 G1 F3000 Z-18*44
Recv: !! Move out of range: 284.324 308.901 -18.000 [60.128]
Changing monitoring state from "Resuming" to "Cancelling"
Recv: ok
Send: N97 M108*20
Send: N98 G91*32
Recv: // Unknown command:"M108"
Recv: ok
Changing monitoring state from "Cancelling" to "Printing"
Recv: ok
Send: N99 M84*47
Send: N100 M104 T0 S0*32
Recv: ok
Send: N101 M106 S0*103
Recv: ok
Changing monitoring state from "Printing" to "Operational"
Recv: ok
Send: M105
Recv: ok T0:210.3 /0.0

[klippy.log](https://github.com/KevinOConnor/klipper/files/2992267/klippy.log)
 -->

wizhippo, 2019-03-21:
I am using https://github.com/KevinOConnor/klipper/blob/f2fbe984547c5291ee4cd9df40d4ee0f36ed1fbb/config/example-extras.cfg#L1441 with success.  I know that is not helping with the relative moves you are trying to do but that combined with https://github.com/KevinOConnor/klipper/blob/f2fbe984547c5291ee4cd9df40d4ee0f36ed1fbb/config/sample-macros.cfg#L44 work well for me.

Arksine, 2019-03-22:
It looks like Octoprint is sending a G90 before it sends the move.  This tells Klipper to use absolute coordinates, which is the opposite of what you want to do.  As @wizhippo suggested you can use Klipper's built in pause/resume functionality, just remember that you shouldn't use Octoprint's UI in combination with it (Just use the PAUSE and RESUME gcodes, or macros containing them).

If you decide to use pause_resume I would also suggest updating Octoprint's config.yaml to add `unknownCommandsNeedAck: true` the `serial:` section.  This won't  be necessary in future releases as the next version will detect Klipper firmware and enable it for you.

KevinOConnor, 2019-03-31:
It looks like this issue was resolved (make sure to explicitly set absolute/relative mode in g-code), so I'm going to close this ticket.

-Kevin

closed 2019-03-31 by KevinOConnor