Klipper3d/klipper #1431: Unable to encode: set_digital_out after Fysetc F6 Upgrade

uuzi, 2019-03-21:
I have been using klipper since a few months now and decided to upgrade my trusty old MKS Gen 1.4 Board to the Fysetc F6 in order to get rid of the cable clutter.

I did a git pull and recompiled the firmware and flashed it. so far so good, klipper service could connect to the mcu. Then I edited the printer.cfg in order to get the board fully working with the new pinout. 

Now I'm getting this error:

`Unable to encode: set_digital_out
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready`

I tried recompiling, reflashing..... no luck.

I'm pretty sure there is something wrong with my config, since it worked with the old config just fine (at least the connecting to the mcu part). But after a few hours of search, I can't seem to find any error.

The X- and Y- endstop pins I had to extract out of the board schematic, since the Fysetc F6 does the wiring of the diag pins (which I need for sensorless homing) via a jumper and the documentation for that is, to put it mildly, a bit lacking. 
I'm suspecting that using an analog pin as a digital IN for the x-endstop could have something to do with the problem, but I can't tell for certain. Normally using analog pins as digital should not be a problem, but I'm by no means an expert.

So any help and hints are much appreciated.

THX in advance.

uuzi

[klippy.log](https://github.com/KevinOConnor/klipper/files/2994032/klippy.log)

uuzi, 2019-03-22:
Over night I had an idea:
Could the problem be, that the mcu is trying to communicate with the tmc2130 drivers via SPI?
Because the drivers are not yet installed.

On the other hand, with the old config, the error didn't occur although SPI was enabled then too with no drivers installed.

But maybe there was a commit that enabled that?

trevjonez, 2019-03-22:
probably a good idea to make sure the board is all hooked up before worrying about software errors. here is my config for a full bear mk3 with the F6 and 2130's https://github.com/trevjonez/klipper-configs/blob/master/fysetc_f6_bear.cfg

uuzi, 2019-03-22:
Thank you! I can now confirm that it has something to do with my config file. Your config file works fine once the deprecated lines are removed (most prominently driver_BLANK_TIME_SELECT which does not exist in newer versions of klipper). 
Now I'll just use your config as a starting point and make changes according to my custom build.

closed 2019-03-22 by uuzi

trevjonez, 2019-03-22:
Ah yes. I must not have pushed those changes yet 😀