Klipper3d/klipper #1433: MKS gen L + TMC2208 UART steppers not moving

acbreyer, 2019-03-22:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2994420/klippy.log)

I followed the "[UART This](https://www.instructables.com/id/UART-This-Serial-Control-of-Stepper-Motors-With-th/)!" guide, solder-bridging all 3 Watterott pads together and using a single wire from the UART pin on the driver to pins 63-66 (AUX) on the board.  I'm using the SilentStepProtectors.  

I've made it up to the "Verify stepper motors" section of the config checks, but nothing I do will make the steppers move.  When I try to home an axis, the motors are locked to the axis and only occasionally will I hear what sounds like commutation, but I never see movement until they "fail to home".  The only indicator of issues I've seen is "OpenLoad_A" and "OpenLoad_B" after a G28, but this isn't always guaranteed.  I'd say I'm about 95% confident in all the physical connections.

To troubleshoot, I copied the settings from https://github.com/KevinOConnor/klipper/issues/196 and https://github.com/KevinOConnor/klipper/issues/1365 separately to try to recreate the successes with no change  Is there something else I should try for diagnostics or an obvious mistake in my configuration?

FHeilmann, 2019-03-22:
You seem to be getting good communication with the steppers. Are you confident in your pin assignments for step/dir and enable signals? 

One thing that stands out is that the drivers report ENN (the enable signal) to be high (ENN=1 in your log) which, as far as I know, disables the stepper driver, so I'd suggest investigating whether you either chose the wrong enable pin in the config or whether the logic has to be inverted. 

Note that this may also be due to the steppers actually being disabled because they weren't supposed to be enabled when you issued the DUMP_TMC command.

acbreyer, 2019-03-22:
ahhhhhhhh I very frustratedly took everything apart and checked the connections again. I made a soldering error in which I mirrored the picture and left off the STEP and CLK pins instead of leaving off MS1 and MS2. I guess that "95% confidence" was more of a completion mark. Sorry about that :sweat:

closed 2019-03-22 by acbreyer