Klipper3d/klipper #1440: [Feature Request] Deltesian Kinematics

Leapo, 2019-03-22:
Requesting support for Deltesian printers, typically constructed from two towers (XZ Axis) and a moving bed (Y axis).

These require simplified (planer) delta motion for the XZ axis and normal cartesian motion for the Y axis.

KevinOConnor, 2019-03-24:
It's fine if someone wishes to add support for this type of kinematics.  Only Python code would be required (see the existing kinematic code in klippy/kinematics/) - check out the docs for more information: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md#adding-new-kinematics

-Kevin

KevinOConnor, 2019-04-14:
I'm going to mark this as inactive as it doesn't appear it is currently being implemented.

-Kevin

closed 2019-04-14 by KevinOConnor

Leapo, 2019-04-16:
Please re-activate. I'm trying to implement it myself, but having no luck. Need some assistance.

It's looking like some C code will need to be changed in order to make it possible, not just Python.

Leapo, 2019-04-29:
Still waiting... Should I make a new feature request and link to this one, instead of requesting this one be re-opened?

KevinOConnor, 2019-05-01:
If you have code ready to submit then open a github pull request with the code.  If you have questions on the code, then go ahead and ask them here.

-Kevin

Leapo, 2020-02-12:
Haven't gotten very far because I'm fairly certain it will require updating more than _just_ Python code. Neither kin_cart.c or kin_delta.c is appropriate for this printer (Cartesian Y axis, planer XZ axis).

Updating the Python I could handle, updating C I'll need help with.

KevinOConnor, 2020-02-12:
Thanks for the update.  I don't consider this closed - I just don't consider it as an "active open issue" either.  Hence the (perhaps misleading) "inactive" tag.  I'm leery of moving this back to the open state because it is then an ongoing maintenance cost for me when I review all the open tickets.  If you have a better suggestion then let me know.

It shouldn't be needed to update the C code.  If I understand it correctly, you can use the kin_cart.c for the Y axis (via a call to `rail.setup_itersolve('cartesian_stepper_alloc', 'y')`) and use the existing kin_delta.c for the other two axes (via a call to `rail.setup_itersolve('delta_stepper_alloc', arm_length**2, tower_x_pos, 0.0)`).  Said another way, the C code is only used to implement the "inverse kinematics" math formula for each stepper and there is already a C implementation of these math formulas.

-Kevin

Leapo, 2020-04-27:
Is there any other example of calling two different sets of kinematics simultaneously? Every time I look at doing this, that's where I get stuck.

Also, would the existing Delta kinematics even work, considering they were designed for 3 towers rather than 2?

KevinOConnor, 2020-04-29:
> Is there any other example of calling two different sets of kinematics simultaneously?

Yes, both the polar and corexy kinematics do that.

> would the existing Delta kinematics even work

I think you're right - I think Deltesian would need its own C based math formula, as the cartesian y coordinate must not contribute to the calculated stepper position for the two towers.

FYI, performance is not an issue at all though.

-Kevin

Leapo, 2021-03-15:
Been a while, figured I ping this thread. I'm still hoping I might find someone to help with this.

I've tried adding the new machine type multiple times, without success. I'm going to need a hand here :(

Tircown, 2021-08-09:
Following a discussion on Klipper's discord with notmktlr, I am implementing this kinematics in Klipper. Code wise it's not a problem but I'd be happy to get feedback on the subtleties of the deltesien kinematics and have people test it.