Klipper3d/klipper #1441: Hotend Thermistor calibration issue with CR10S

wilsonkg, 2019-03-23:
Hi Kevin, I just want to start by saying that Klipper is awesome !
There  - now that we have that out of the way.
I'm not having a problem with Klipper exactly but what I am experiencing is that the reported temp and the actual temp of my hotend don't match and I would like to correct that.
I have a CR10S and the sensor_type is EPCOS 100K B57560G104F.
The nozzle is actually 17c cooler than the reported temp.
Can I add something to the config file to correct this or do I have to alter a py file and if so could you direct me to the correct location?

Thank you and keep up the fantastic work.

wilsonkg, 2019-03-24:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2999941/klippy.log.zip)

jakep82, 2019-03-24:
You can define a custom thermistor.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L1260

KevinOConnor, 2019-03-31:
As @jakep82 indicates, if the thermistor is different, one can provide the actual thermistor parameters.  Also, if one is willing to measure a series of actual temperature/resistance measurements then one can use an `[adc_temperature my_sensor]` config section with those measurements to provide a more accurate thermistor for a particular printer.

-Kevin

KevinOConnor, 2019-04-14:
I'm going to close this as it appears the conversation has concluded.

-Kevin

closed 2019-04-14 by KevinOConnor