Klipper3d/klipper #1442: Filament Insert Trigger goes of randomly

shaiss, 2019-03-24:
I think there's a bug in the filament_switch_sensor code but I'm not sure why it's happening.  My printer will be sitting idle (or even printing, but it's the same error), then there will be a "switch_sensor: insert event detected, Time 13279.80 " per the below snippet from the klippy.log.  I've added a pullup sensor with ^ to the config, but no luck and same thing.  Note that `QUERY_FILAMENT_SENSOR SENSOR=e0_filament` does report filament as expected.

Another point too, why is the filament_sensor even active if the printer isn't printing?  I believe the even should only fire if the printer isn't printing.

I can see the insert event kicking off when not printing, but in that case, I'd prefer to manually trigger the "load_filament" macro I have, vs klipper kicking of an insert event.

snipete of error line 2737:
`
[klippy-filament.log](https://github.com/KevinOConnor/klipper/files/2999951/klippy-filament.log)
[klipper-localcopy.txt](https://github.com/KevinOConnor/klipper/files/2999952/klipper-localcopy.txt)

Must home axis first: 60.000 0.000 0.034 [0.000]
Must home axis first: 60.000 0.000 0.034 [0.000]
Script running error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/filament_switch_sensor.py", line 62, in _exec_gcode
    self.gcode.run_script(script)
  File "/home/pi/klipper/klippy/gcode.py", line 285, in run_script
    res = self.process_batch(commands)
  File "/home/pi/klipper/klippy/gcode.py", line 265, in process_batch
    self.process_commands(commands, need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 195, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 75, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 37, in cmd
    self.gcode.run_script_from_command(script)
  File "/home/pi/klipper/klippy/gcode.py", line 278, in run_script_from_command
    self.process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 195, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 75, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/gcode.py", line 514, in cmd_G1
    raise error(str(e))
error: Must home axis first: 60.000 0.000 0.034 [0.000]
Stats 13280.7: gcodein=1373 mcu: mcu_awake=0.009 mcu_task_avg=0.000096 mcu_task_stddev=0.000079 bytes_write=11220 bytes_read=142783 bytes_retransmit=0 bytes_invalid=0 send_seq=1428 receive_seq=1428 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15998364 heater_bed: target=0 temp=19.2 pwm=0.000 print_time=13260.269 buffer_time=0.358 print_stall=0 extruder: target=0 temp=19.4 pwm=0.000
`

Arksine, 2019-03-25:
The runout event is active while printing, the insert event is active when the printer is in the ready or idle state.   The insert event cannot trigger while the printer is in the "printing" state.

Looking at your log it doesn't appear that the insert detection is happening while printing.  If its going off randomly I would suggest checking your wiring and your sensor mount.  Make sure that the filament completely depresses the switch.  Its possible that the switch itself is malfunctioning.

If you prefer to manually trigger inserts vs using the sensor then you can remove the insert_gcode option.  This will disable insert detection.   However I would still suggest checking your switch and wiring, if something is loose then you will likely trigger random runout events as well.

shaiss, 2019-03-27:
Yeah it's odd. I ran 7+ he's of prints without 1 false alert.  Actually no alerts since the filament never really ran out.

LazaroFilm, 2019-03-28:
That filament runout thing is hard to test. It's like the airbags of your car.

KevinOConnor, 2019-04-14:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-04-14 by KevinOConnor

shaiss, 2019-04-14:
Yeah that's fine. Somehow it's now stable and not doing this anymore.
Maybe it was the trip to #mrrf