Klipper3d/klipper #1444: Max extruder acceleration

Paulbsh, 2019-03-24:
Hi folks,
even setting the retraction speed to 80mm/s, my extruder is retracting very slowly.
I guess it's due to the overall acceleration set to 700 (very heavy bed in a i3 scenario).
Is there a way to set the E acceleration independently like it can be done with Z?

Thanks

jakep82, 2019-03-24:
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L120

Paulbsh, 2019-03-24:
As this is not an actual issue, I didnt attach the klippy.log. 
Here it is
[klippy.log](https://github.com/KevinOConnor/klipper/files/3000766/klippy.log)
For completeness, this is my actual config:
https://pastebin.com/z0XDiHQx?fbclid=IwAR1Sm4EkJZoTn2uk1N27-dLt1jN99cJui9J1fO7Qr40akR-92KdvMpVbHX4

hevilp, 2019-03-29:
Thought about that, It is just a Feature request

jakep82, 2019-03-29:
I've already provided a link to the answer.  Yes, this can be done by simply adding a single line to your extruder config.

Paulbsh, 2019-03-31:
> I've already provided a link to the answer. Yes, this can be done by simply adding a single line to your extruder config.

yes, that did work. Just forgot to say thanks :)
Must have overlooked that setting in the sample files

closed 2019-03-31 by Paulbsh