Klipper3d/klipper #1445: Intersecting infill cross hatch lines don't connect

drphil3d, 2019-03-24:
I've been struggling with a strange issue I keep running into. It's almost as if the printer is not extruding enough material as it intersects other crosshatch lines. 
My extruder is calibrated correctly, yet this occurs with or without pressure advance enabled. Low vs High print speed seems to have little affect. I experience this issue the most when using a cross hatch infill, however it seems to be an issue with all the infill patterns.

Exterior Print quality is good but parts are weakened.
It sometimes helps when I use a thicker infill line thickness. Currently set to 0.45

[image]

[image]

[image]

[klippy playV2 infill issues.log.txt](https://github.com/KevinOConnor/klipper/files/3000847/klippy.playV2.infill.issues.log.txt)

Hywelmartin, 2019-03-25:
In my opinion does this looks like under extrusion due to too fast printing(not adequate melting). So bump up the temperature and/or slow down the infill..

megablue, 2019-03-25:
are you using cura?

drphil3d, 2019-03-25:
Yes I am using Cura 3.6 and 4.0

bshowket, 2019-03-25:
I also had same issue, i couldn't fix this issue using cura 3.6. I moved to ideamaker since and didn't see any issue since.

drphil3d, 2019-03-26:
I don't seem to have the same issue as much when using Slic3r

megablue, 2019-03-27:
it is a problem with cura combing. it doesn't account for the molten filament oozing away during the combing moves. set the combing mode to **Not in Skin** and Max Comb Distance to somewhere between 20-50mm.

drphil3d, 2019-03-27:
yeah I made sure my extruder was perfectly calibrated, still have the same issue. even over extruding it does it. 

That makes sense I will try that.

drphil3d, 2019-03-27:
Now that I think about it a little more it makes perfect sense. The further I get away from the 1st layer with an infill pattern the worse it gets because it’s slowly loosing material over the duration of the print. I could see the extruder gears slow down and stop when ever crossing paths with another infill line. However it would occur inconsistently all over the print, some objects worse then others. It’s been driving me absolutely crazy I thought well it must be the crazy experimental firmware, but why am I the only one having this problem. 🥺

drphil3d, 2019-03-27:
@megablue I did as you suggested ... and it got worse 
[image]

drphil3d, 2019-03-27:
I've switched to using Slic3r instead of Cura, but I'm getting the same exact problem. Changing Print Temp/Speed/Extrusion/Max Accel/Etc have little to no effect.

Setting max velocity lower makes it easier to see the problem
 SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500 and SET_PRESSURE_ADVANCE 
ADVANCE_LOOKAHEAD_TIME=0

The drawing attached shows where I can watch the extruder shaft stop moving right before it's about to intersect and leaves this half connected line.
[square.gcode.txt](https://github.com/KevinOConnor/klipper/files/3011484/square.gcode.txt)

Its only doing 1 retract at the start of the next layer
[image]

with travel moves visible in green
[image]

[klippy3-26-2019.log](https://github.com/KevinOConnor/klipper/files/3011478/klippy3-26-2019.log)
[image]

Hywelmartin, 2019-03-27:
infill overlap, I use 20-25% and you'r using 10%..

If you like I could test to print the same at your settings..(I updated 3 days ago).
what infill, layer height and so on are you using..

drphil3d, 2019-03-27:
Infill overlap controls the overlap between the infill and the inner shell.
Setting this at 20% or 0% seems to have no affect.
The default profile for a printrbot simple or play in cura works. 0.2 mm
layer height.
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

Arksine, 2019-03-27:
Does it do it with Pressure Advance off?

drphil3d, 2019-03-29:
I feel like a real asshole here, but It seem to me like the root of the problem is to do with the nozzle width being set to a larger value then what it physically.

closed 2019-03-29 by drphil3d