Klipper3d/klipper #1448: Pin mapping

schnello, 2019-03-25:
Hello.

Machine: Any i3 Mega
FW: Lastest Klipper Build

I installed an BLtouch in my machine. It works but want know what I did wrong.
Regarding some guides and other custom FW (Marlin) the servo pin should be:
#define SERVO0_PIN      **11**

[image]

I match also with my machine:
[image]

In Klipper i can only move the sensor if i set it to:
control_pin: ar5

It looks like the pinout for this block is mirrored
4,5,6,11,12 vs 12,11,6,5,4

So why it works in Marlin with pin 11 and in Klipper with pin 5? I do not think all the guides are wrong but I have not even tested it under Marlin.

PS: pin_map is set to arduino

Hywelmartin, 2019-03-25:
I have the same board but in a delta.
I use those pins for my TMC-2130 as CS-pins and at first I used 4,5 and 6..for A,B and C and after a while I also used pin 11 for E so on my board it is as the picture.. 

you could easily test it with a simple sketch like Blink...

schnello, 2019-03-25:
> also used pin 11 for E so on my board it is as the picture..
> you could easily test it with a simple sketch like Blink...

Hmm strange. I testet it several times and on my board is D11 ==> D5.

schnello, 2019-03-25:
Hmm. I compile now the Marlin FW and there is the same "issue". the second from the right is pin5 (and not 11). So this is not Klipper releated i closed the ticket. If anyone reads this ticket and have information regarding the reason: "Why the servo pinout is mirrored" feel free to contact me.

closed 2019-03-25 by schnello

Hywelmartin, 2019-03-25:
The D11 is PB5.. are you sure that you don't have a typo... or you could try to use PB5 to see if there's a difference.

schnello, 2019-03-25:
As i wrote i test this also with Marlin. Same result: Pin5 is OK and works with the pins installed on the photo. Yes the Klipper config is: control_pin: ar5 

Marlin Config:
#define SERVO0_PIN      5

So iam not a Ardunino Guru but from my postion ... the pin is 5.

 >or you could try to use PB5
So instead of ar5 ... pb5?

Hywelmartin, 2019-03-25:
I'm a little curious because I have the same card.. have you tried all ports 4,5,6,11 and 12

yes you can use the hardware pin number PB5 but it would render the same result.... I'm afraid...

[image]

schnello, 2019-03-25:
>have you tried all ports 4,5,6,11 and 12

I try to install the bltouch exactly after the instructions. After 2 days i try the pin nr 4,5 ... and i stop at 5 because it was successfully. So at the end iam happy that it works but i want also understand why the pin is 5 and not 11. :)

I test also the HW PIN. Same result: 
PB5 failed
PE3 good