Klipper3d/klipper #1450: set_gcode_offset error with dual carriage (IDEX) printer

w1ebr, 2019-03-26:
Hi Kevin,

I am trying to use both extruders on my IDEX printer (first time). I noticed a bug in the handling of "set_gcode_offset". I have offsets defined for both extruders for X and Z. If you look at my "enter_dumpzone" macro, the problem behavior I am seeing is when the Y only move takes place, "extruder" moves to the left by about 18 mm. Note that the 18.5 mm offset is associated with "extruder1".

I haven't found a work-around yet, so this is keeping me from using both extruders in one print.

Thanks, Kevin.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3006283/klippy.log)

KevinOConnor, 2019-03-31:
I guess you are reporting that a Y only move after a T0/T1 is resulting in X movement.  This is the intended behaviour of the SET_GCODE_OFFSET command.

If you don't want that X movement, you should be able to accomplish something similar by replacing the SET_GCODE_OFFSET command with the M206 command.

FWIW, given issue #1289 we'll likely need to change some of the behaviour of the SET_GCODE_OFFSET command.  I don't have any time frame to do that, however.

-Kevin

w1ebr, 2019-04-01:
Thanks, Kevin. I think I misunderstood the meaning of SET_GCODE_OFFSET.

closed 2019-04-01 by w1ebr