Klipper3d/klipper #1451: Add support for ST7565 LCD

gavinmetzler, 2019-03-26:
I have an AZSMZ Mini + 12864 LCD (ST7565). Support for the LCD panel would be great.

KevinOConnor, 2019-03-27:
It's fine if someone wishes to add support for this.  I'd guess the developer would need direct access to the hardware.  (It's not something I personally will have time to look at.)

FWIW, it looks like this display is very similar to the uc1701 display - it may require just a different init sequence.

-Kevin

terrorhai, 2019-04-12:
Hey, here is the link to the ST7567 pinout (which should be very similar to the 65) and how it's integrated in Marlin. I'm not a programmer, but eventually it helps for anyone to integrate that display type. I would be very happy, if someone makes a pull request :)
https://wiki.fysetc.com/Mini12864_Panel/

condemil, 2019-04-13:
Is someone else is using st7567 LCD with RGB leds around encoder here is the configuration for ramps-compatible board, it works on MKS Gen L too.

```ini
[display]
lcd_type: st7567
rs_pin: ar23
contrast: 63
cs_pin: ar17
a0_pin: ar16
encoder_pins: ^ar31, ^ar33
click_pin: ^!ar35

[output_pin red]
pin: ar25
value: 1

[output_pin green]
pin: ar27
value: 1

[output_pin blue]
pin: ar29
value: 1
```

terrorhai, 2019-04-18:
>Is someone else is using st7567 LCD with RGB leds around encoder here is the configuration for ramps-compatible board, it works on MKS Gen L too. ...

@condemil thank you so much, works super!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

condemil, 2019-04-22:
Also for FYSETC Mini12864 Panel users - you need to desolder the following components to prevent "Can not update MCU 'mcu' config as it is shutdown" and "Failed automated reset of MCU 'mcu'" errors during `FIRMWARE_RESTART` command:

C4 and R4 to have reset on panel button
OR
C4 and R3 to have kill on panel button though ar41 pin

KevinOConnor, 2019-04-28:
I'm going to close this as it looks like support is now available.

Thanks,
-Kevin

closed 2019-04-28 by KevinOConnor