Klipper3d/klipper #1452: Klipper with BIGTREETECH SKR 1.3

vekenti, 2019-03-26:
Hello. I'm having troubles flashing klipper on the new SKR 1.3 board. No matter what I do it does not work. Even doing a completely fresh install of raspbian and klipper did not work.

I believe I got klipper to flash successfully onto the board, but something in the code is screwing up. Heres the error that I am receiving:

> tail /tmp/klippy.log
>     self._check_config()
>   File "/home/pi/klipper/klippy/mcu.py", line 583, in _check_config
>     self._send_config(None)
>   File "/home/pi/klipper/klippy/mcu.py", line 557, in _send_config
>     self._serial.send(c)
>   File "/home/pi/klipper/klippy/serialhdl.py", line 131, in send
>     cmd = self.msgparser.create_command(msg)
>   File "/home/pi/klipper/klippy/msgproto.py", line 331, in create_command
>     raise error("Unable to encode: %s" % (msgname,))
> error: Unable to encode: config_soft_pwm_out

Someone else on the klipper discord also is having problems with flashing klipper. I'm not sure if it is due to the same error or not as I haven't gotten in touch with him yet.
 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3008700/klippy.log)

jakep82, 2019-03-26:
PA4 is not a valid pin name in your bltouch section.

vekenti, 2019-03-26:
That fixed it. Thank you so much.

closed 2019-03-26 by vekenti