Klipper3d/klipper #1455: Delta arm length

novio8, 2019-03-27:
Hi,
I just started Klipper install in Ubuntu. I have Delta printer and I wonder how
cruical Delta arm length Value is, having in mind difficulty measuring the length
precisely.

Regards
Anton

Hywelmartin, 2019-03-27:
A measuring tape is good enough as a starting point plus minus 2 mm.. delta callibration and enhanced delta calibration will probably sort the rest... If you have been using the printer with some other FW those values should be good as well

novio8, 2019-03-27:
thank you,
I have Repetier Firmware installed, and I have been printing with my Values
with Calibration turned off and I get prints that are dimensionally very acurate
Tolerances are around 0.1 mm

closed 2019-03-29 by novio8