Klipper3d/klipper #1456: Ubunu install

novio8, 2019-03-27:
Yesterday I installed Ubuntu and Octoprint. Both work OK.
Today I used Klipper script for Ubuntu and installation went through.
I can see Klipper runing in Octoprint Terminal but I can't Create 3D printer
Firmware and flash it to my printer.
Anyone able to help?
Thank you

voodoo-bravo, 2019-03-27:
Are you using ubuntu 18.04? 
This might be related https://github.com/KevinOConnor/klipper/issues/443

novio8, 2019-03-27:
yes I have the latest of Ubuntu 18.04.2. But I did not try to Flash Klipper
firmware to Arduino, because I don't know how. I'm new to Linux OS

novio8, 2019-03-28:
Today I managed to finally flash Klipper firmware. On Ubuntu
installer folder name is klipper-master, and not klipper as it is named for Pi
So I just executed commands as are written in installation.md

At the I can report that Klipper is running on latest Ubuntu 18.04.2

novio8, 2019-03-29:
Klipper is running on latest Ubuntu 18.04.2, made few prints
on my Delta style printer
Thanks Kevin for good Software

closed 2019-03-29 by novio8