Klipper3d/klipper #1457: Layers shifting with external drivers, missed steps

gavinmetzler, 2019-03-27:
On my delta printer with Leadshine ND556 drivers, some steps are being missed somehow leading to shifting layers - the print comes out sort of OK, but wonky.

I can confirm steps are missed because after a print, before homing again, manually commanding printer to a known spot on the table (X0 Y0), it doesn't return to the same spot, plus it would hit the table if sent to Z0.

I think it is due to timing between a DIR change and the next Step pulse. I have seen others comment about this, but I didn't see any solution. I see an option to change the step pulse length was added, and I have tried 2, 5 and 8 - there seems to be some difference when I change it, but the problem isn't solved.

It would be great to have a setting for a minimum delay between DIR and next STEP, or even just a confirmation from the developers of what this timing actually is on a LPC1769. The ND556 driver docs specify a minimum of 5us.

The external drivers are driven through a SN74HCT245N transceiver to convert the 3.3V from the LPC1769 to 5V. (I actually think the cheap AZSMZ board uses an LPC1768 and overclocks it at 120mhz)

If you care to read my whole story, its in a reply post on the smoothieware forum titled "External driver: constant step lost" (http://forum.smoothieware.org/forum/t-9350867/external-driver:constant-step-lost#post-4179333)

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3015421/klippy.zip)

KevinOConnor, 2019-03-28:
The step delay specified in "make menuconfig" also applies to DIR changes.  (Technically, a DIR change could be slightly less time, but a setting of 8us should ensure at least a 7us DIR time.)

If you think the chip is actually an LPC1768, then select LPC1768 during "make menuconfig".  (It wont harm anything - it will just cause Klipper to configure the clocks at a slightly slower speed.)

Alas, I don't have any guesses as to why your drivers are skipping steps.

-Kevin

ajanky, 2019-03-30:
You should check if changing the step pulse polarity helps (/step) - some drivers trigger on the rising and others on the falling edge - that means the active edge can appear before the dir change, resulting in the described problem.

gavinmetzler, 2019-03-31:
Thanks for the help. Being able to confirm that the time between DIR change and next STEP is definitely more than the driver's spec (5us) enabled me to eliminate some possibilities and I decided the ND556 drivers are likely faulty.

I did try inverting the Step pin too - that lead to drastic layer shifting (and I confirmed from the driver docs that it does step on rising edge, so inversion isn't needed)

I have gone back to stepstick drivers (drv8825) which are noisy and running near the limit of their voltage/current (40V/2A) but its working, no layer shifting!

gavinmetzler, 2019-03-31:
> 
> 
> The step delay specified in "make menuconfig" also applies to DIR changes. (Technically, a DIR change could be slightly less time, but a setting of 8us should ensure at least a 7us DIR time on the LPC176x.)
> 
> If you think the chip is actually an LPC1768, then select LPC1768 during "make menuconfig". (It wont harm anything - it will just cause Klipper to configure the clocks at a slightly slower speed.)
> 
> Alas, I don't have any guesses as to why your drivers are skipping steps.
> 
> -Kevin

Hi Kevin,

I just wanted to clarify - did you mean that the length of the DIR pulse is set by the delay setting in "make menuconfig", or the time delay between DIR and next STEP? 

I used a cheap logic analyser to look at the signals and it seems most of the time, the time between a DIR change and next STEP is huge, but every now and then it would be a lot closer, but still nowhere near the 5us minimum required by the driver.

But then again, the prints usually come out almost right - only a few steps are missed out of millions, so it still worries me that maybe every now and then the firmware is doing a STEP after DIR change in a short (<5us) time - is this possible?

Its not urgent anymore because I've changed drivers and the prints are coming out good now. I haven't tried bumping up the speed to what I was able to with external drivers, but accuracy is more important than speed right now, but I'd still like to know for sure if my $500 worth of drivers are useless, or if they are just missing steps because of timing issues from the firmware...

KevinOConnor, 2019-03-31:
The "step pulse duration" in "make menuconfig" sets the step pulse high duration, step pulse low duration, and lead time for DIR change to step pulse.  The minimum DIR time is twice the "step pulse duration".

Specifically, if a "step pulse duration" is configured then Klipper sets the DIR signal immediately after the step pulse low of the previous step.  Thus, the minimum DIR lead time is also equal to the minimum step pulse low time, which is equal to the configured "step pulse duration".  (Technically, as noted before, the "minimum DIR time before step pulse" is slightly less time than the "step pulse duration" because it technically occurs after the step pulse low, but this time delta should always be less than 1us on an lpc176x micro-controller.)

FWIW, you could try upping the time to 15us just to see if maybe the driver spec is not accurate.

-Kevin

KevinOConnor, 2019-04-14:
I'm going to close this as it appears the conversation has concluded.  I'll reopen if there are further insights.

-Kevin

closed 2019-04-14 by KevinOConnor