Klipper3d/klipper #1460: Klipper state: Shutdown when performing a long "Z" axis move

jalanjarosz, 2019-03-28:
Piper 2 CoreXY with a 300mm x 300mm x 400mm build volume
[klippy.log](https://github.com/KevinOConnor/klipper/files/3017355/klippy.log)

MCU = AT Mega with RAMPS 1.6+ and TMC2130 for X, Y, and E axis
MCU Z = AT Mega with a CNC Sheild and DRV8825 at 16 micro-steps for Z axis

Z-Rods are T8*2 single start

X and Y moves are all good, but when sending a G1 x250, the MCU shuts down with:

06:15:53
MCU 'z' shutdown: Timer too close

Klippy.log attached.

Line 1376 is a G1 X250
Line 1390 is a G1 X150
Line 1400 is a G1 Z250

These are manual moves, no streamed G-Code involved.  The printer is currently being assembled.

If I send multiple G1 Z??? commands at 100, 200, 300, 390, the axis will go up without issue. It's only when sending larger Z values.

There is another Piper community member who is using a KFB3.0 board with the same issue.

jalanjarosz, 2019-03-28:
Host is a Raspberry Pi 3B+ running a clean and latest install of Octoprint.  The Pi is running headless (no graphical UI).

KevinOConnor, 2019-03-30:
This is a known issue with Klipper on machines with very precise stepper motors - the processing time for the host machine to generate all the steps for a very long move may be greater than the time allocated to it.  In your particular case, a Z move of 250mm would require 1.6million steps to be generated, and it is taking the RPi longer over 400ms to generate those steps.

Unfortunately, there is no simple work around - possibilities solutions include reducing the z axis micro-steps, reducing the available z axis travel distance, manually breaking up long z moves into multiple z moves (eg, G1 Z50; G1 Z100; G1 Z150; G1 Z200; G1 Z250), and installing Klipper on a more powerful (desktop class) machine.  There is a long term plan to fix this in Klipper, but there is currently no time frame for that.

-Kevin

jalanjarosz, 2019-03-30:
@KevinOConnor -
Thank you for taking the time to explain this limitation. I can use the multi-step work-around for now.

If you have a master ticket this issue is reported in, feel free to link this issue to it.  You may also leave it as an open issue or close it.  Which ever is appropriate for your tracking purposes.

Best regards.
-James

vladbabii, 2019-09-23:
@KevinOConnor  i think this could be solved easier by
1) we have a lookahead buffer option to generate moves X moves / gcode bytes / time in advance and set that to 10 second or 30 second

2) have an option to generate a "kcode" file with all the needed moves from a gcode file, save it, then run that instead of gcode so at print time we only stream mcu commands

As far as i remember from reading the docs i think there's a way to emulate running a gcode file to test things out. Can we modify that to produce klipper move files?

Would any of these options be realistically doable ?

closed 2019-11-07 by KevinOConnor

KevinOConnor, 2019-11-07:
This should be fixed as of PR #2130 .

-Kevin

PR #2130 (2019-11-07): Rework step generation to use a C based trapezoidal velocity queue
This series is a large backend rework of the Klipper kinematics.  The current code generates the steps for each move, one move at a time.  This pull request changes the underlying step generation code to generate steps based on a time range.  Each host stepper object is associated with a list of moves, and steps are then generated from the moves on that list.  A step generation time range may be part of a move, multiple moves, or some combination.

This change should fix #1460.  It will also make it much easier to merging PR #1 [... cut]