Klipper3d/klipper #1464: Bed mesh leveling features

danielkucera, 2019-03-29:
Hi,

I want to discuss some possible improvements for automatic bed mesh leveling.
Marlin supports following options: http://marlinfw.org/docs/gcode/G029-abl.html
I'd love to see support for L,R,F,B parameters so I've made some code changes with some results but I'm not sure if it is worth to continue.
Do you consider this useful? 
In case I develop a proper pull request would you be willing to include such feature?

bruce356, 2019-03-30:
@danielkucera Hi, not sure what you are after, thought this was taken care of by:-
#Probed area below = 250mm x 320mm
min_point: 11,45	;The minimum start point (x,y) on the grid to be generated
max_point: 241,342	;The maximum end point (x,y) on the grid to be generated

The dimensions take into account the BLtouch offsets from hot end.
Regards - bruce
L,R,F,B left right front back

danielkucera, 2019-04-01:
Hi @bruce356 , I understand that this can by set in klipper config but I'd like to set the area dynamically. So the leveled area will correspond to the size of the current model. This has an advantage that if you do the same number of probes while utilizing a smaller area, you will get better accuracy and higher speed. One usually doesn't use the whole printbed area for printing.

KevinOConnor, 2019-04-01:
It's not really clear to me what you are proposing, so it's hard for me to provide any suggestions.

Cheers,
-Kevin

PeggyFree, 2019-04-02:
Hello, 

@danielkucera , instead of this feature, I think it would be more efficient to set the G28 Z homing at the center of the model . I think doing a G29 at each print is overkill, while Z homing at the right place would significantly improve reliability on large printers.

danielkucera, 2019-04-02:
Hi @PeggyFree , this is already happening with my printer. Why do you think G29 is overkill to do every time? There are many factors which can cause different bed level...

KevinOConnor, 2019-04-21:
I'm going to close this as it looks like the discussion has concluded.

-Kevin

closed 2019-04-21 by KevinOConnor