Klipper3d/klipper #1466: Klipper print side lines

novio8, 2019-03-30:
My Delta printer printed just fine with Repetier Firmware,
Yesterday I started printing using Klipper and visible side lines
are printed on my models.
Any ideas what is the cause ?
[image]

novio8, 2019-03-31:
I attached log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/3027076/klippy.log)

KevinOConnor, 2019-03-31:
It's pretty common to get results like that when using "control: watermark" in the [heater_bed] section.  If possible, follow the directions at https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md#calibrate-pid-settings to run PID calibration on heater_bed.

-Kevin

novio8, 2019-04-01:
KevinOConnor, 
I followed your directions and my prints again as expected.
Thank you

closed 2019-04-01 by novio8