Klipper3d/klipper #1467: Pause -->  disconneceted Octoprint but print doesen't stop..

Hywelmartin, 2019-03-30:
So I printed(v-SD) and saw a booger on my hotend. So to wipe it of I used my Pause_park macro..
It took a while to get executed so I triggered it again before. so I got multiple pauses.. And the print head didn't Park like it should..(I have to test this more)
After a while i noticed that Octoprint got disconnected and the print went on happily, so I tried to reconnect but that didn't work... for some time... then it connected and the print stopped.. sort of..
The print head just stopped and heaters was still on at set temperatures, motor was engaged. Fire hazard?? the Idle timer would probably turn everything off..

I believe that this is mostly a octoprint issue. Will  it  be fixed by the upcoming release that adress "unknownCommandsNeedAck: true"? 
I have edited the config.yaml
```
serial:
  abortHeatupOnCancel: false
  additionalPorts:
  - /tmp/printer
  autoconnect: true
  baudrate: 250000
  disconnectOnErrors: false
  port: /tmp/printer
  timeout:
    temperature: 9.0
    temperatureAutoreport: 3.0
    temperatureTargetSet: 6.0
    unknownCommandsNeedAck: true
```

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/3025304/klippy.log.zip)

Hywelmartin, 2019-03-31:
Tested RC1 of Octoprint and same result
without unknownCommandsNeedAck: true

Arksine, 2019-03-31:
How are you triggering the pause?  Via the menu?

Hywelmartin, 2019-03-31:
Via macros.. that uses Pause/resume that is set up in FW
I disabled 
[image]
And I was able to get reconnected.. once..

Hywelmartin, 2019-03-31:
another log where octoprint timed out, but I got reconnected after a while..

[klippy190331.zip](https://github.com/KevinOConnor/klipper/files/3026696/klippy190331.zip)

in this one I believe that the heaters timed out and I did set them waited and then resume.. didn't go so good..

Arksine, 2019-03-31:
I don't think the issue is with acknowledgements. 

The log shows that you tried to resume when the printer wasn't paused, and it also shows that you tried to Pause when it was already paused (or in the process of pausing).   Its possible that previously you paused without a subsequent resume, and havent restarted since.   Have you tried testing after restarting Klipper?

The other thing that could cause this behavior is attempting to Pause the SD card via the display/menu, which as you say you are not doing.

Arksine, 2019-03-31:
I just looked at your latest log.  I wonder if you are somehow overwhelming the psuedo-tty with all the calls to SET_VELOCITY_LIMIT.  Maybe there is something going wrong on octoprint's side.

Hywelmartin, 2019-03-31:
I only get this behavior after I have used the pause/resume function...

Arksine, 2019-04-01:
Okay, lets clarify your workflow so I can get a better idea of what is going on.

1)  You are printing via virtual_sdcard, monitoring with Octoprint
2)  You use Octoprint's terminal to enter a PAUSE gcode, or a macro containing the PAUSE.  You are NOT using Octoprint's built in Pause and Resume buttons.
3) The printer keeps printing, regardless of the pause command

Verify for me that the above is correct.  It appears that pause_resume thinks that you are printing from octoprint, not from virtual_sd.   There is something about your setup that I am missing, as I cannot reproduce your issue.

Hywelmartin, 2019-04-01:
I ended the tests early..(#3)
indeed it is connected to chitchat of SET_VELOCITY_LIMIT xxxx.

I have been printing a while with SET_VELOCITY_LIMIT instead of M204 without any problems..

[klippy_M204_first_second_set_veloc....zip](https://github.com/KevinOConnor/klipper/files/3030782/klippy_M204_first_second_set_veloc.zip)

closed 2019-04-02 by Hywelmartin