Klipper3d/klipper #1468: stm32f0: cross talk between ADC channels

smark-, 2019-03-30:
Hi Klipper developers,

the stm32f0 ADC channels seem to have some cross talk, in particular, when I start heating the tool, the bed's temperature reading is affected. I'd like to point out that this is not an hardware issue since everything works fine on the v0.7.0.
[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/3025742/klippy.log)

KevinOConnor, 2019-03-31:
You appear to be using an stm32f103 chip.  A similar issue was reported in #1261.  Perhaps @yschroeder, @grigorig, or @nowiresza have more info.

-Kevin

yschroeder, 2019-03-31:
I am not an expert on the ADC, but @smark- could test the values reported in #1261. If they fix the problem here as well, then we should consider using them as default.

smark-, 2019-03-31:
Yes! The solution proposed in #1261 works. I'd be happy to make a pull request, since I believe it's currently a bug to be fixed.

KevinOConnor, 2019-03-31:
Yes, I agree it is a bug, and it would help if a Pull Request is created for it.

Thanks,
-Kevin

nowiresza, 2019-03-31:
Yes #1261 has been working very well for me. Now more ADC cross talk issues to date. I've actually use sampling times down to LL_ADC_SAMPLINGTIME_41CYCLES_5 and had now negative effects.

smark-, 2019-03-31:
Sure, I'm on it

closed 2019-03-31 by smark-