Klipper3d/klipper #1470: lpc176x with smoothie bootloader: switching to bootloader not seen by the kernel

pulsar256, 2019-03-30:
Board: `Big Tree Tech SKR 1.3, stock bootloader`
Kernel: `Linux 5.0.5-arch1-1-ARCH`
Applies to: 8c7f1546ab1d43ddf6d839467df16fb01c6f3269
FollowUp to: https://github.com/KevinOConnor/klipper/pull/1344

When the lpc176x microcontroller is put into the "bootloader" aka DFU mode (via 1200 Baud request), the new operating mode is not seen by the operating system / kernel. It still assumes the device to operate in ACM mode. To work around this state, a USBDEVFS_RESET ioctl can be issued after which the device gets re-recognized with a new vendor and productId and can be flashed using dfu_util.

Ideally the klipper should be able to trigger a disconnect / reconnect procedure seen by the kernel when leaving the ACM mode. It works the other way around right after DFU-flashing is complete.

Workaround / modified `flash_usb.py` (needs root permissions)

```python
# Flash via a call to dfu-util
def flash_dfuutil(device, binfile, extra_flags=[], sudo=True):
    hexfmt_r = re.compile(r"^[a-fA-F0-9]{4}:[a-fA-F0-9]{4}$")
    if hexfmt_r.match(device.strip()):
        call_dfuutil(["-d", ","+device.strip()] + extra_flags, binfile, sudo)
        return
    buspath, devpath = translate_serial_to_usb_path(device)
    enter_bootloader(device)
    pathname = wait_path(devpath)
    
    # HACK: make this optional if lsusb does not report the expected 
    # vendor / product id?
    reset_usb_device() 
    
    call_dfuutil(["-p", buspath] + extra_flags, binfile, sudo)

def reset_usb_device():
    proc = subprocess.Popen(["lsusb"], stdout=subprocess.PIPE)
    out = proc.communicate()[0]
    lines = out.split('\n')
    for line in lines:
        # TODO: consider looking for vendor/productId instead. 
        # productId is user-configurable though.
        if 'Arduino SA' in line:
            parts = line.split()
            bus = parts[1]
            dev = parts[3][:3]

    device_path = "/dev/bus/usb/%s/%s" % (bus,dev)
    sys.stderr.write("Resetting device %s\n" % (device_path))
    fd = os.open(device_path, os.O_WRONLY)
    try:
        fcntl.ioctl(fd,21780,0) # 21780 = USBDEVFS_RESET
    except IOError as e:
        pass # we expect an io error as the path will disappear 
    finally:
        os.close(fd)
    # TODO: do not guess, wait until the device comes back up 
    time.sleep(5) 
```

If it turns out, that this particular issue cannot be fixed in the firmware, I can offer to finish / clean up the work above and open a PR.

KevinOConnor, 2019-03-31:
That's odd.  What OS/host machine are you using?

Can you attach the klipper log file from a working session (so we can see the software settings/config).  Then could you run `lsusb`, `make flash FLASH_DEVICE=/dev/ttyACM0`, then `lsusb`.  Cut-and-paste the full terminal output here.

Thanks,
-Kevin

pulsar256, 2019-03-31:
Thanks for looking into it.

Running:

* OS: Arch Linux 
* Kernel: x86_64 Linux 5.0.5-arch1-1-ARCH

# pre-flash state
```
~/develop/klipper (master ✘)✭ ᐅ ls /dev/serial/by-id -lah
total 0
lrwxrwxrwx 1 root root 13 Mar 31 10:26 usb-Klipper_Klipper_firmware_12345-if00 -> ../../ttyACM0

~/develop/klipper (master ✘)✭ ᐅ lsusb | grep Arduino
Bus 001 Device 024: ID 2341:abcd Arduino SA 
```

# flashing 
```
~/develop/klipper (master ✘)✭ ᐅ make flash FLASH_DEVICE=/dev/ttyACM0
  Flashing out/klipper.bin to /dev/ttyACM0
Entering bootloader on /dev/ttyACM0
Device reconnect on /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-7/1-7:1.0
sudo dfu-util -p 1-7 -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

Failed to flash to /dev/ttyACM0: Error running dfu-util

If flashing Klipper to a Smoothieboard for the first time it may be
necessary to manually place the board into "bootloader mode" - press
and hold the "Play button" and then press and release the "Reset
button".

When a Smoothieboard is in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1d50:6015

Alternatively, one can flash a Smoothieboard via SD card - copy the
"out/klipper.bin" file to a file named "firmware.bin" on an SD card
and then restart the Smoothieboard with that SD card.

make: *** [src/lpc176x/Makefile:49: flash] Error 255
```

# post-flash state

```
~/develop/klipper (master ✘)✭ ᐅ lsusb | grep Arduino                
Bus 001 Device 024: ID 2341:abcd Arduino SA 
~/develop/klipper (master ✘)✭ ᐅ lsusb | grep "\(Arduino\|Moko\)"
Bus 001 Device 024: ID 2341:abcd Arduino SA 
```

# Manually resetting
```
~/develop/klipper (master ✘)✭ ᐅ sudo resetusb /dev/bus/usb/001/024
Resetting USB device /dev/bus/usb/001/024
Error in ioctl: No such device
~/develop/klipper (master ✘)✭ ᐅ lsusb | grep "\(Arduino\|Moko\)"  
Bus 001 Device 025: ID 1d50:6015 OpenMoko, Inc. Smoothieboard
~/develop/klipper (master ✘)✭ ᐅ 
```

 `make flash FLASH_DEVICE=1d50:6015` would work now

# device reset, flashing with workaround

```
~/develop/klipper (master ✘)✭ ᐅ git stash pop
(...)
	modified:   scripts/flash_usb.py
(...)
~/develop/klipper (master ✘)✹✭ ᐅ 

~/develop/klipper (master ✘)✹✭ ᐅ sudo make flash FLASH_DEVICE=/dev/ttyACM0
  Flashing out/klipper.bin to /dev/ttyACM0
Entering bootloader on /dev/ttyACM0
Device reconnect on /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-7/1-7:1.0
Resetting device /dev/bus/usb/001/026
sudo dfu-util -p 1-7 -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6015
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 512
Copying data from PC to DFU device
Download	[=========================] 100%        19108 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
~/develop/klipper (master ✘)✹✭ ᐅ lsusb | grep "\(Arduino\|Moko\)"         
Bus 001 Device 028: ID 2341:abcd Arduino SA 
```

KevinOConnor, 2019-03-31:
Thanks.  It really would help to have the klipper log file from the software during a normal run.

Does the following patch improve things for you?
```diff
--- a/src/lpc176x/usbserial.c
+++ b/src/lpc176x/usbserial.c
@@ -251,6 +251,7 @@ usb_request_bootloader(void)
     // The "LPC17xx-DFU-Bootloader" will enter the bootloader if the
     // watchdog timeout flag is set.
     irq_disable();
+    sie_cmd_write(SIE_CMD_SET_DEVICE_STATUS, 0);
     LPC_WDT->WDMOD = 0x07;
     NVIC_SystemReset();
 }
```

If it does not improve things, does the following patch improve things?
```diff
--- a/src/lpc176x/usbserial.c
+++ b/src/lpc176x/usbserial.c
@@ -8,6 +8,7 @@
 #include "LPC17xx.h" // LPC_SC
 #include "autoconf.h" // CONFIG_SMOOTHIEWARE_BOOTLOADER
 #include "board/irq.h" // irq_disable
+#include "board/misc.h" // timer_read_time
 #include "byteorder.h" // cpu_to_le32
 #include "command.h" // output
 #include "generic/usb_cdc.h" // usb_notify_ep0
@@ -251,6 +252,10 @@ usb_request_bootloader(void)
     // The "LPC17xx-DFU-Bootloader" will enter the bootloader if the
     // watchdog timeout flag is set.
     irq_disable();
+    sie_cmd_write(SIE_CMD_SET_DEVICE_STATUS, 0);
+    uint32_t end = timer_read_time() + timer_from_us(5000);
+    while (timer_is_before(timer_read_time(), end))
+        ;
     LPC_WDT->WDMOD = 0x07;
     NVIC_SystemReset();
 }
```

-Kevin

pulsar256, 2019-03-31:
* patch1: no effect
* patch2: :+1: did the trick.

Here are the logfiles of the normal operation before applying patch1 or patch2 if that still matters:

```
Starting Klippy...
Args: ['klippy/klippy.py', './skr1.3-ender3.cfg', '-v', '-l', './klipper.log', '-I', 'klipper.sock']
Git version: 'v0.7.0-375-g75b73b0-dirty'
CPU: 12 core AMD Ryzen 5 2600 Six-Core Processor
Python: '2.7.16 (default, Mar 11 2019, 18:59:25) \n[GCC 8.2.1 20181127]'
Start printer at Sun Mar 31 16:10:10 2019 (1554041410.3 21336.9)
===== Config file =====
[tmc2208 stepper_x]
uart_pin = P1.17
microsteps = 16
interpolate = true
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 250

[stepper_x]
step_pin = P2.2
dir_pin = !P2.6
enable_pin = !P2.1
step_distance = .0125
endstop_pin = P1.29
position_endstop = 0
position_max = 320
homing_speed = 50

[stepper_y]
step_pin = P0.19
dir_pin = !P0.20
enable_pin = !P2.8
step_distance = .0125
endstop_pin = P1.27
position_endstop = 0
position_max = 300
homing_speed = 50

[stepper_z]
step_pin = P0.22
dir_pin = P2.11
enable_pin = !P0.21
step_distance = .0025
endstop_pin = P1.25
position_endstop = 0.5
position_max = 400

[extruder]
step_pin = P2.13
dir_pin = !P0.11
enable_pin = !P2.12
step_distance = .010526
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = P2.4
sensor_type = EPCOS 100K B57560G104F
sensor_pin = P0.24
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 260

[heater_bed]
heater_pin = P2.5
sensor_type = ATC Semitec 104GT-2
[... 108 lines cut ...]
Stats 21375.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=3983 bytes_read=8106 bytes_retransmit=9 bytes_invalid=0 send_seq=217 receive_seq=217 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002324 heater_bed: target=0 temp=22.8 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.7 pwm=0.000
Stats 21376.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=3989 bytes_read=8226 bytes_retransmit=9 bytes_invalid=0 send_seq=218 receive_seq=218 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002377 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.6 pwm=0.000
Stats 21377.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=3995 bytes_read=8347 bytes_retransmit=9 bytes_invalid=0 send_seq=219 receive_seq=219 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002384 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.7 pwm=0.000
Stats 21378.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=4001 bytes_read=8468 bytes_retransmit=9 bytes_invalid=0 send_seq=220 receive_seq=220 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002394 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.7 pwm=0.000
Stats 21379.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=4007 bytes_read=8574 bytes_retransmit=9 bytes_invalid=0 send_seq=221 receive_seq=221 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002330 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.7 pwm=0.000
Stats 21380.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=4013 bytes_read=8695 bytes_retransmit=9 bytes_invalid=0 send_seq=222 receive_seq=222 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002405 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.7 pwm=0.000
new minimum rtt 21381.426: hrtt=0.000058 freq=100002405
Stats 21381.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=4019 bytes_read=8830 bytes_retransmit=9 bytes_invalid=0 send_seq=223 receive_seq=223 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002320 heater_bed: target=0 temp=22.8 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.6 pwm=0.000
Stats 21382.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=4025 bytes_read=8936 bytes_retransmit=9 bytes_invalid=0 send_seq=224 receive_seq=224 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002272 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.6 pwm=0.000
Stats 21383.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=4031 bytes_read=9057 bytes_retransmit=9 bytes_invalid=0 send_seq=225 receive_seq=225 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002212 heater_bed: target=0 temp=22.9 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.6 pwm=0.000
[... cut ...] me=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.6 pwm=0.000
Stats 21398.9: gcodein=0 mcu: mcu_awake=0.001 mcu_task_avg=0.000007 mcu_task_stddev=0.000006 bytes_write=4121 bytes_read=10839 bytes_retransmit=9 bytes_invalid=0 send_seq=240 receive_seq=240 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=100002233 heater_bed: target=0 temp=22.8 pwm=0.000 print_time=13.341 buffer_time=0.000 print_stall=0 extruder: target=0 temp=21.6 pwm=0.000```

KevinOConnor, 2019-03-31:
Okay, thanks.  Commit fbbe2cf9.

-Kevin

pulsar256, 2019-03-31:
Validated against d1eefba4536055fa99bc872bd2a64c9076407398 - working just fine. Closing the issue. Thanks!

closed 2019-03-31 by pulsar256