Klipper3d/klipper #1472: Mesh levelling over compensates in corrections.

dmp450, 2019-03-31:
**Problem:**
Mesh levelling seems to be overcompensating in its corrections. The center of my bed is calibrated properly. However, when I try to print a bed level test, the nozzle seems to be too far away on the right side, and too close on the left side. I am using an authentic BLTouch, and an MKS Gen L. I have verified:

* My offsets are correct;
* my zsteps are correct;
* the z motors are actually turning while printing;
* My axis are all square and level.

Attached are copies of my log and my printer config.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3026780/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3026781/printer.cfg.txt)

Arksine, 2019-03-31:
Looking at your configuration something isn't right with it.  Your x_offset in [probe] is set to -21.5.  Your min_point in [bed_mesh] is 20,50.  This would put the probe off of the bed, as it would be 1.5mm to the left of X=0.

dmp450, 2019-03-31:
Yeah, I caught this after I posted. The cause for that is because my endstop doesn't let my nozzle go right to the edge of the bed. Either way, I adjusted that x minimum point to be 25 and am still getting the same results. As a test, I also disabled mesh levelling to see how my print was. Everything basically reversed. The sides that were too close were too far, and vice versa.

Arksine, 2019-03-31:
That makes sense looking at the points in your configuration.  Without adjustment the right side will be too low, the left too high. 

Looking at the points in your config It seems like something is causing your probe to trigger too high on the right and too low on the left. If you are printing with heat, its possible that your bed geometry is changing.  You could try pre-heating the bed, doing an automatic probe, then doing a first layer test.

dmp450, 2019-04-01:
Just tried it, and preheating did not solve things. The results are the same.

Arksine, 2019-04-01:
You could try probing the mesh using manual probing.  I suspect that something is causing the probe to lose its relationship with the nozzle at the extents of the x range.   Its also possible that you are experiencing some backlash.

dmp450, 2019-04-01:
I'll give that a shot. If that's the case, would it possibly fix it if I decrease my x range further? This leads to my next question: is there possibly something wrong with the bed mesh code if the solution is to do a manual probing?

Also, I'm sorry, but I'm not familiar with backlash.

Arksine, 2019-04-01:
The code for bed_mesh is well tested and works on many printers.  I have reviewed the code many times attempting to fix issues others have had, I can't find anything wrong with it or reproduce the issues others face.  Typically it comes down to a configuration issue or a hardware issue.  In the end the mesh is only as good as the input it receives.  If manual probe works then most likely there is an issue with your probe, either it isn't consistent or the relationship between the probe and the nozzle is changing.

You can try decreasing the range on X as well, if there is something affecting travel at the extents it may help.

Backlash is exhibited when an axis travels further in one direction than another due to slack in the system.  For example, you might command the head to move Z up 10mm and it does indeed move up 10mm.  Then you command the head to move down 10mm, only it moves down 9.8mm.

dmp450, 2019-04-01:
I'll try out manual and see if it solves problems. I'll also try out a friend's BLTouch to make sure mine isn't the problem. If needed, I suppose I could try to figure out Marlin and see if I get those issues over there. I'm really enjoying Klipper otherwise, so I'd like to continue using it. This will take a few days to do; is it okay if this issue remains open until I report back?

Arksine, 2019-04-02:
I'm sure Kevin will leave it open long enough for you to do some tests and get back.  I definitely would suggest trying manual first.  If you try Marlin with success, please record all the results from the mesh (The probe coordinates, the probed z positions, and the generated mesh).

FWIW, Marlin has a featured called "Mesh Editing" that allows users to change points that aren't probed accurately (they are no strangers to users having issues with mesh leveling).  I have plans to implement something similar in bed_mesh at some point, although I don't have a timetable on it.

dmp450, 2019-04-04:
For an update: I've tried manual probing, and I've tried using a new BLTouch, fresh out of the package. The problem is *exactly* the same. I'll try Marlin this weekend and provide a final update.

dmp450, 2019-04-05:
Final update: Marlin gives the exact same problem. This must mean:

* I'm consistently making the same mistake (possible)
* The location of my printer is in the middle of some gravitational or electromagnetic anomaly (unlikely)
* Something is wrong

closed 2019-04-05 by dmp450

Arksine, 2019-04-05:
Thanks for the update.  It could be any number of things, but if it were me the first thing I would do is to try swapping the the drivers for the Z axis.   

Given that you are running independent drivers for each Z motor, you may want to have a look at the z_tilt module as well.