Klipper3d/klipper #1473: Display generates running text

AndyWo83, 2019-03-31:
Hello,
today i have problems with my Display on Ender 3 with MKS GenL Board.
The display generates "running" text. 
When the print is finish and I restart the host or the FW, the display is ok.
This failure is only when I print a thing or I heat up my nozzle or bed.

Today after this prolem I installed Klipper new, but nothing changed...
[klippy.log](https://github.com/KevinOConnor/klipper/files/3026861/klippy.log)
[image]

KevinOConnor, 2019-04-01:
The description sounds like the result of electrical noise being picked up by the lcd wires.  Double check that the plugs are secure and the wiring looks okay.  It may also be useful to try and separate the lcd wires from other wiring (such as stepper motors, heaters, mains power).

-Kevin

AndyWo83, 2019-04-01:
Hi Kevin,
the wiring has been like this for 3 months since I chance to MKS Gen L. 
It started all at once and does not leave. 
But I will relocate the cable differently to see if it is gone then.

AndyWo83, 2019-04-04:
So, I have chance the Display flat wire and the text run again...
Now I have removed the Display and place it to the side and now the error is gone!
Does anyone have a tip why could this have an impact?

AxMod3DPrint, 2019-04-04:
@AndyWo83 - wrap the ribbon cables in aluminium foil then tape it on to stop shorts, this should help with the interference

KevinOConnor, 2019-04-14:
I'm going to close this as it appears a solution was found for the original issue.  (Alas, these lcd displays are very sensitive to noise - not much we can do in software.)

-Kevin

closed 2019-04-14 by KevinOConnor