Klipper3d/klipper #1475: Change to different analog port than the 3 thermistor ports

BladeScraper-Designs, 2019-03-31:
So... I managed to mess up my mainboard (MKS Gen L).  Seems like 2 of the 3 thermistor ports are fried.. bed one reads fine, but the other two read 250C and 128C respectively.  I was wondering if it's possible to use a spare analog port to act as the analog input for the thermistor until I'm able to get a new board.  I was thinking about using a few on Aux2, but it seems like they're dual purpose.. they're labeled A11/D65 and A12/D66.  Is there a way for me to use one of these as a thermitor input?  If I just put A11 into the config, it says it's not a valid analog input.

Hywelmartin, 2019-03-31:
you need a 4.7k pullup resistor

KevinOConnor, 2019-04-11:
As the bot indicates, we need the klipper log file in order to assist.

-Kevin

closed 2019-04-11 by KevinOConnor