Klipper3d/klipper #1477: ADC out of range despite hardware working/correct config

DMCShep, 2019-03-31:
I am getting an incessant issue with "ADC out of range". This only occurred when I upgraded my mainboard from a ReliaBuild board (pre-included on RigidBots) to an MKS Gen v1.4 and also swapped the heated bed for a Keenovo mains 600W silicone heated bed with thermistor, all other hardware remained identical. From searching this seems to be related to thermistor configs being wrong -- which I copied verbatim from another MKS Gen v1.4 user (plus my own for the thermistor type) -- or the hardware being bad -- which worked fine before and nothing changed on that front so that wouldn't be it either.

The temps are reading at zero for the extruder and nonexistent for the bed, as seen here:

[image]

There is also a TON of noise in the klippy.log file about UART -- I am using the header wiring and it is colored
```
Red | Orange | Yellow | Brown
 X  |    Y   |    Z   |   E
```
[image]
Pins should be correct for the MKS Gen v1.4 mainboard:

[image]

Printer config:
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3027196/printer.cfg.txt)

Full log file (as much as it would accept anyhow):
[klippy_2019-03-31.log](https://github.com/KevinOConnor/klipper/files/3027195/klippy_2019-03-31.log)

Any ideas what's causing the ADC out of range issue? And is there an underlying UART problem also? If so, from which stepper? I can't seem to figure that out or if it's even a problem at all.

Thank you!

jakep82, 2019-03-31:
The sensor pins on your extruder and bed are wrong. I believe they should be analog13 and analog14.

DMCShep, 2019-03-31:
I'm confused, isn't that what they are set to?

Relevant sections:
```
[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 245

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: pid
pid_Kp: 70.936
pid_Ki: 2.056
pid_Kd: 611.821
min_temp: 0
max_temp: 130
```

jakep82, 2019-03-31:
I only looked at the top of your log which shows ar67 and ar68.

DMCShep, 2019-03-31:
Bad thermistor caused this -- hardware was faulty after all, disregard ticket

closed 2019-03-31 by DMCShep