Klipper3d/klipper #1482: Multiple extruder bed mesh compensation

sgtnoodle, 2019-04-01:
This is an issue originally brought up in the mailing list.

Klipper may not properly deal with mesh based bed leveling when using multiple extruders. The config file doesn't have anywhere to specify an offset from the first nozzle to the second. I assume the firmware needs to know the offset, though, so that on a tool change the z height will be appropriately adjusted based on the bed mesh.

Based on some Google searches, it sounds like SET_GCODE_OFFSET combined with an "activate_gcode" section in the config file might work, but it isn't clear to me whether or not the gcode offset will also offset the mesh based z height adjustment.

@KevinOConnor:
> I suspect you are correct and that multiple extruders may not use the
> correct probe offset when using bed mesh.  However, depending on
> whether or not the mesh is accounting for deviations in the rail vs
> deviations in the bed - it may or may not be correct to use a
> different probe offset.

Arksine, 2019-04-01:
Yes, you can use SET_GCODE_OFFSET to accommodate the difference between different nozzles.  The mesh offset is applied after the gcode offset, so they "stack".  The activate gcode would likely be the correct place to put this.

KevinOConnor, 2019-04-01:
I think the issue is with the probe offset.  Lets take an example:
at extruder0 position 0,0 / probe position 10,0 -> the height is measured at 0.0
at extruder0 position 10,0 / probe position 20,0 -> the height is measured at 0.1
at extruder0 position 20,0 / probe position 30,0 -> the height is measured at 0.0

Then, if there is a second extruder which is nominally 20mm positive X from the first extruder, it would be:
at extruder1 position 20,0 / extruder0 position 0,0 / probe position 10,0 -> height adjustment = ?
at extruder1 position 30,0 / extruder0 position 10,0 / probe position 20,0 -> height adjustment = ?
at extruder1 position 40,0 / extruder0 position 20,0 / probe position 30,0 -> height adjustment = ?

I'm not sure the current code would do the "right thing".  I'm not even sure what the "right thing" is, though.

-Kevin

sgtnoodle, 2019-04-01:
Re: deviations in the rail vs. deviations in the bed. Since most probes are offset from the nozzle but on the same carriage, I think rail deviation is inherently difficult to compensate for. With the offset applied, the carriage is moved to a different place along the rail than where the height was probed. Since that's the case, it seems like offsetting for the first nozzle is just as bad as offsetting for multiple nozzles. It works well when height error is due to bed deviations, and works poorly when height error is due to rail distortions. If someone really cared about rail distortions, it seems like they would need to use a zero offset probe. For multiple nozzles, they would need a zero offset probe for each nozzle.

Based on that, it seems like bed distortion is what matters when applying probe offsets. The current behavior of applying the offsets before applying the mesh correction seems less than ideal for dealing with bed distortion. It would be better to apply the mesh offsets based on the probe's position, and then add the extruder offset on top of that.

Currently, the config file allows you to specify a probe offset. Ideally, one would be able to specify per-extruder offsets from extruder to probe. I suppose this could be done in the existing schema by setting the probe offset to 0 and then using "activate_gcode" and SET_GCODE_OFFSET on each extruder (unless activate_gcode isn't supported on the main extruder).

Arksine, 2019-04-02:
@KevinOConnor  Its possible that could present a problem.   Would it be correct to assume that a gcode offset is applied to x and y when a tool change is complete?  If that is the case, I believe bed_mesh will look up the correct z value.  Using your example:

Extruder1 position 0,0 =  Extruder0 position 20,0 / probe position 30,0 -> height measured at 0.0
Extruder1 position 10,0 = Extruder0 position 30,0 / probe position 40,0 -> height measured at 0.1

Of course I could be wrong on this.  I'll research multi-extruder setups and how tool changes are performed to get a deeper understanding of what is happening.

@sgtnoodle Indeed, mesh leveling cannot account for issues with the rest of the printer's geometry.  If there is some kind of distortion in the rail that significantly changes the relationship between the nozzle and the bed over the probe offset's range then I agree that a zero offset probe would be required.   

I'm not sure I'm following your suggestion about applying offsets based on the probe's position.  The offsets are used to translate nozzle coordinates to the probe's coordinates during the mesh generation phase.  This maps the mesh directly to the area of the bed that was probed.  Once this is done the offsets are no longer used.

KevinOConnor, 2019-04-02:
@Arksine - My understanding of the current code (I could be wrong) is:

With:

extruder0 position 0,0 / probe position 10,0 -> the height is measured at 0.0
extruder0 position 10,0 / probe position 20,0 -> the height is measured at 0.1
extruder0 position 20,0 / probe position 30,0 -> the height is measured at 0.0

the probe offset would be 10,0 then and it would result in a mesh that moved the Z up .1mm when extruder0 is at position 20,0. So, for extruder0 moves, we'd get something like:

extruder0 position 0,0 / probe position 10,0 -> height adjustment = 0.0
extruder0 position 10,0 / probe position 20,0 -> height adjustment = 0.0
extruder0 position 20,0 / probe position 30,0 -> height adjustment = 0.1

That is, we'd correctly move the extruder0 up a little when going over the bump at position 20,0.

Then, if extruder1 is activated, it would use a SET_GCODE_OFFSET X=-20.  Thus, when the user commanded a logical movement from X0 -> X1, it would effectively move extruder0 from X-20 to X-19 (which would ideally result in extruder1 moving from X0 -> X1).  After this gcode offset, then the results would be:

extruder1 position 20,0 / extruder0 position 0,0 / probe position 10,0 -> height adjustment = 0
extruder1 position 30,0 / extruder0 position 10,0 / probe position 20,0 -> height adjustment = 0
extruder1 position 40,0 / extruder0 position 20,0 / probe position 30,0 -> height adjustment = .1

I'm not sure what's desired, as it could depend of how extruder1 is mounted and the type of distortion measured.  However, bumping up extruder1 when it is at position 40,0 does seem odd.

Again, I could be wrong in my understanding.
-Kevin

Arksine, 2019-04-02:
@KevinOConnor I'm following now.  I agree, this is going to be a problem.   If Extruder1 is at X=40 in real space, bed_mesh will receive X=20.   This is going to result in looking up the coordinate where Extruder0 is, not Extruder1.

I think removing the offset from the coordinate prior to looking up Z would solve this.  I could add a gcode to the mesh that allows the user to set a "tool offset" and add that to the coordinate prior to the z lookup. 

```MESH_TOOL_OFFSET X=20 Y=0```

Not sure I like that gcode name, but essentially give the mesh an offset relative to Extruder0.

sgtnoodle, 2019-04-05:
Just brainstorming a bit. The best workaround I can think of in the existing implementation would be to create two bed meshes, one for each extruder's offset from the probe. Then, the extruders' activate_gcode sections could use BED_MESH_PROFILE to load the appropriate profile on a tool change. I don't know if those two bed meshes could be created at runtime via scripting, or if it would require config file changes, but it would at least adjust properly.

I like the idea behind `MESH_TOOL_OFFSET`. I'd suggest supporting Z in addition to X and Y just in case. One slight variant would be to call it just `TOOL_OFFSET` and then have it both offset the gcode and the mesh compensation appropriately. I personally would prefer if it were config parameters rather than gcode commands, but that's philosophical. I don't know why Klipper is evolving toward more and more procedural gcode routines, but maybe there's good reasons for it. 

```
[extruder]
    tool_offset: 0,0,0

[extruder1]
    tool_offset: 0,0,0
```
or
```
[extruder]
    activate_gcode:
        TOOL_OFFSET  ; Omitted axes default to 0

[extruder1]
    activate_gcode:
        TOOL_OFFSET x=10 y=10  ; z defaults to 0
```

megablue, 2019-04-28:
i like this idea a lot i am facing a similar dilemma where other toolheads of my printers are forced to use SET_GCODE_OFFSET for XYZ offsets adjustment after toolchange. since SET_GCODE_OFFSET is used, if there i messed up the bed leveling, i no longer able to use SET_GCODE_OFFSET to adjust the z height during a print. having tool_offset on top of gcode_offset is definitely really convenient.  

i second the idea to support Z offset for tool as well, since it is always an issue for some multi hotend setup to align the nozzle to the correct height, and with some hotend it is literally impossible without additional physical adjustment mechanism.

KevinOConnor, 2021-06-09:
This issue should be resolved with the merge of #4346.

-Kevin

closed 2021-06-17 by github-actions[bot]