Klipper3d/klipper #1483: Original BLTouch "BLTouch failed to verify sensor state"

dmagyar, 2019-04-01:
I have an Ender 3 with an MKS Gen L v1.0 board fitted. I was using Marlin until yesterday with my original BL Touch happily for months. I decided to give Klipper a try and I have to say: VERY impressed!

Unfortunately when trying G28 Z it gives me the error "BLTouch failed to verify sensor state" :( I have not changed my wiring and it works perfectly if I flash back my Marlin. Checked with the DEBUG statements I can deploy/stow the pin of the touch, it self-tests OK on bootup etc.

Attaching my log generated with M112. Please find my printer.cfg below (yes display is not working yet, unrelated to this issue):

```
[stepper_x]
step_pin: PF0
dir_pin: !PF1
enable_pin: !PD7
step_distance: .0125
endstop_pin: ^PE5
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PF6
dir_pin: !PF7
enable_pin: !PF2
step_distance: .0125
endstop_pin: ^PJ1
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PL3
dir_pin: PL1
enable_pin: !PK0
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_min: -2
position_max: 250

[extruder]
step_pin: PA4
dir_pin: !PA6
enable_pin: !PA2
step_distance: .010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PK7
control: pid
pid_Kp: 31.71
pid_Ki: 3.69
pid_Kd: 68.18
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PH5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PK6
control: pid
pid_Kp: 460.67
pid_Ki: 90.70
pid_Kd: 584.93
min_temp: 0
max_temp: 130

[fan]
pin: PH6

[mcu]
serial: /dev/ttyUSB0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[bltouch]
control_pin: PB5
sensor_pin: ^PD3
x_offset: -49
y_offset: -10
z_offset: -1.43

[homing_override]
gcode:
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X166 Y120 F6000 ; Change the X and Y coordinates to the center of your print bed
    G28 Z
set_position_z: 0.0

[bed_mesh]
speed: 100
horizontal_move_z: 5
samples: 2
sample_retract_dist: 3
min_point: 78,15
max_point: 230,210
probe_count: 3,3

#[display]
#lcd_type: st7920
#cs_pin: PA3
#sclk_pin: PA3
#sid_pin: PC1
#encoder_pins: ^PH0, ^PA1
#click_pin: ^!PC2
```
[klippy.log](https://github.com/KevinOConnor/klipper/files/3031164/klippy.log)

KevinOConnor, 2019-04-01:
If you're sure that the wiring is correct, you can add a "pin_up_touch_mode_reports_triggered: False" to the bltouch section to disable that sensor check.

-Kevin

sgtnoodle, 2019-04-02:
I am just now setting up a clone BLTouch sensor. I get that error too. Yes yes it's not supported :-). Using manual test commands, though, I see that the endstop reads TRIGGERED when commanding pin_up followed by touch_mode. Perhaps there needs to be some settling time added somewhere?

I have no idea what touch_mode actually means. Do you have a good reference document for the BLTouch sensor? I downloaded the "manual" but it doesn't explain what anything actually means.

skykinetic, 2019-04-02:
Does anyone else happen to be running a genuine BLTouch v2.2 as well? I am too facing the same issue that required `pin_up_touch_mode_reports_triggered: False` to get the probe working. 

From the official website, v2.2 was released only in Jan 2019 and the minor increment in number seems to reflect a firmware change. Judging from all the search results I found online, v3.0 users seems to be also facing the same issue on Marlin (see this reddit [post](https://www.reddit.com/r/ender3/comments/b59soj/psa_new_bltouch_v30_will_not_work_with_marlin_11x/)), but I cannot find any referring to v2.2 although marlin v2.0.x did issue a fix. 

My hypothesis is that BLTouch v2.2 introduced firmware changes similar to that of v3.0 and would require similar workaround. Do enlighten me if I'm wrong.

KevinOConnor, 2019-04-03:
FWIW, I've also seen reports that BLTouch v3 requires `pin_up_touch_mode_reports_triggered: False` - so we may need to update the documentation to be more clear.

-Kevin

mtsgsd, 2019-04-18:
@KevinOConnor Antclabs has a statement saying that the V3 probes may need the mainboard to be modified in order to work correctly. [https://www.antclabs.com/wiring3 ](url) I've had one user on our discord that did need to do this.

rnelias, 2019-05-12:
@dmagyar how have you built your printer.cfg? I also have an Ender 3 with MKS Gen L v1.0 with TMC2288 and E3DV6 extruder's clone from trianglelabs and I'm now trying to make it work with klipper firmware but with no success as I can't configure my steppers and extruder :(

KevinOConnor, 2019-05-21:
FYI, BL-Touch documentation updates for this issue are pending in PR #1661.

-Kevin

KevinOConnor, 2019-05-26:
FYI, the documentation has been updated.

-Kevin

closed 2019-06-02 by KevinOConnor