Klipper3d/klipper #1484: kinematics request

manu7irl, 2019-04-01:
Hi I am going through the kinematics docs and I cannot make sure if kinematics for delta rotary robot could be really close to the standard delta one. Or maybe it needs a new kind of kinematics?
Could someone tell me if delta config is good enough for that?
https://youtu.be/6oefrWdS_3M
https://youtu.be/nhq8V75JYZs
https://www.thingiverse.com/thing:3489886
thanks a lot.
Emanuel.

KevinOConnor, 2019-04-01:
As far as I know, rotary delta kinematics are different from linear delta kinematics, and thus some new code would be required.  There's documentation on adding kinematics at: https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md#adding-new-kinematics .  The cable winch commit may also be a good reference: 5dc74f31.

-Kevin

KevinOConnor, 2019-04-21:
I'm going to close this issue as it looks like the discussion has concluded.

-Kevin

closed 2019-04-21 by KevinOConnor