Klipper3d/klipper #1485: M109 -1 out of range

mentaluproar, 2019-04-02:
Simplify3D throws:
M109 T1 S-1 
to disable tool 1 heater.
Klipper responds with
!! Requested temperature (-1.0) out of range (0.0:250.0)

How am I supposed to get my slicer to communicate this with Klipper properly?

----log below----
Starting Klippy...
Args: ['/home/octoprint/klipper/klippy/klippy.py', '/home/octoprint/printer.cfg', '-l', '/var/log/klippy.log']
Git version: 'v0.7.0-349-gd766587'
CPU: 4 core ARMv7 Processor rev 5 (v7l)
Python: '2.7.16 (default, Mar 15 2019, 13:03:02) \n[GCC 8.2.1 20181127]'
Start printer at Tue Apr  2 01:39:16 2019 (1554169156.6 758.4)
===== Config file =====
[stepper_x]
step_pin = ar54
dir_pin = ar55
enable_pin = !ar38
step_distance = .0125
endstop_pin = ^ar3
position_endstop = 0
position_max = 240
homing_speed = 50

[stepper_y]
step_pin = ar60
dir_pin = ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^ar14
position_endstop = 0
position_max = 245
homing_speed = 50

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .00125
endstop_pin = ^!ar18
position_endstop = -2.6
position_min = -2.6
position_max = 175

[extruder]
step_pin = ar26
dir_pin = ar28
enable_pin = !ar24
step_distance = .0052
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar10
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog13
control = pid
pid_kp = 28.310
pid_ki = 1.832
pid_kd = 109.346
min_temp = 0
max_temp = 250

[extruder1]
step_pin = ar36
dir_pin = ar34
enable_pin = !ar30
step_distance = .0052
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar7
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog15
control = pid
pid_kp = 30.092
pid_ki = 1.715
pid_kd = 132.027
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control = pid
pid_kp = 72.125
pid_ki = 2.558
pid_kd = 508.483
min_temp = 0
max_temp = 130

[fan]
pin = ar9

[mcu]
serial = /dev/ttyUSB0
pin_map = arduino

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100
=======================
Extruder max_extrude_ratio=0.266081
Extruder max_extrude_ratio=0.266081
Starting serial connectgot {'#receive_time': 759.7166196530001, u'next_clock': 3856072704L, u'oid': 14, u'value': 7763, '#name': u'analog_in_state', '#sent_time': 759.685870403}
got {'#receive_time': 759.746604819, u'next_clock': 3856552704L, u'oid': 17, u'value': 7762, '#name': u'analog_in_state', '#sent_time': 759.7381694439999}
got {'#receive_time': 759.928603153, u'next_clock': 3859432704L, u'oid': 5, u'value': 7844, '#name': u'analog_in_state', '#sent_time': 759.891372278}
got {'#receive_time': 760.019600445, u'next_clock': 3860872704L, u'oid': 14, u'value': 7757, '#name': u'analog_in_state', '#sent_time': 759.9934220279999}
got {'#receive_time': 760.0496698610001, u'next_clock': 3861352704L, u'oid': 17, u'value': 7756, '#name': u'analog_in_state', '#sent_time': 760.0446266949999}
Loaded MCU 'mcu' 77 commands (v0.7.0-349-gd766587-20190320_035638-KOS-MOS / gcc: (GCC) 8.3.0 binutils: (GNU Binutils) 2.32)
MCU 'mcu' config: RECEIVE_WINDOW=192 SOFT_PWM_MAX=256 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 CLOCK_FREQ=16000000 MCU=atmega2560 STATS_SUMSQ_B>
Configured MCU 'mcu' (622 moves)
Starting heater checks for heater_bed
Starting heater checks for extruder
Starting heater checks for extruder1
Stats 760.2: gcodein=0 mcu: mcu_awake=0.023 mcu_task_avg=0.000135 mcu_task_stddev=0.000077 bytes_write=743 bytes_read=3413 bytes_retransmit=9 bytes_invalid=0 se>
Stats 761.2: gcodein=0 mcu: mcu_awake=0.023 mcu_task_avg=0.000135 mcu_task_stddev=0.000077 bytes_write=749 bytes_read=3429 bytes_retransmit=9 bytes_invalid=0 se>
Stats 762.2: gcodein=0 mcu: mcu_awake=0.023 mcu_task_avg=0.000135 mcu_task_stddev=0.000077 bytes_write=755 bytes_read=3529 bytes_retransmit=9 bytes_invalid=0 se>
Stats 763.2: gcodein=52 mcu: mcu_awake=0.023 mcu_task_avg=0.000135 mcu_task_stddev=0.000077 bytes_write=761 bytes_read=3685 bytes_retransmit=9 bytes_invalid=0 s>
Stats 764.2: gcodein=52 mcu: mcu_awake=0.023 mcu_task_avg=0.000135 mcu_task_stddev=0.000077 bytes_write=767 bytes_read=3841 bytes_retransmit=9 bytes_invalid=0 s>
Stats 765.2: gcodein=52 mcu: mcu_awake=0.023 mcu_task_avg=0.000135 mcu_task_stddev=0.000077 bytes_write=773 bytes_read=3997 bytes_retransmit=9 bytes_invalid=0 s>
Stats 766.2: gcodein=52 mcu: mcu_awake=0.010 mcu_task_avg=0.000088 mcu_task_stddev=0.000069 bytes_write=779 bytes_read=4153 bytes_retransmit=9 bytes_invalid=0 s>
Stats 767.2: gcodein=52 mcu: mcu_awake=0.010 mcu_task_avg=0.000088 mcu_task_stddev=0.000069 bytes_write=785 bytes_read=4309 bytes_retransmit=9 bytes_invalid=0 s>
Stats 768.2: gcodein=57 mcu: mcu_awake=0.010 mcu_task_avg=0.000088 mcu_task_stddev=0.000069 bytes_write=791 bytes_read=4479 bytes_retransmit=9 bytes_invalid=0 s>
Stats 769.2: gcodein=57 mcu: mcu_awake=0.010 mcu_task_avg=0.000088 mcu_task_stddev=0.000069 bytes_write=797 bytes_read=4621 bytes_retransmit=9 bytes_invalid=0 s>
Stats 770.2: gcodein=57 mcu: mcu_awake=0.010 mcu_task_avg=0.000088 mcu_task_stddev=0.000069 bytes_write=803 bytes_read=4777 bytes_retransmit=9 bytes_invalid=0 s>
Stats 771.2: gcodein=57 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=809 bytes_read=4961 bytes_retransmit=9 bytes_invalid=0 s>
Stats 772.2: gcodein=57 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=815 bytes_read=5103 bytes_retransmit=9 bytes_invalid=0 s>
Stats 773.2: gcodein=62 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=821 bytes_read=5259 bytes_retransmit=9 bytes_invalid=0 s>
Stats 774.2: gcodein=62 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=827 bytes_read=5429 bytes_retransmit=9 bytes_invalid=0 s>
Stats 775.2: gcodein=62 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=833 bytes_read=5571 bytes_retransmit=9 bytes_invalid=0 s>
Stats 776.2: gcodein=62 mcu: mcu_awake=0.011 mcu_task_avg=0.000098 mcu_task_stddev=0.000070 bytes_write=839 bytes_read=5741 bytes_retransmit=9 bytes_invalid=0 s>
Stats 777.2: gcodein=62 mcu: mcu_awake=0.011 mcu_task_avg=0.000098 mcu_task_stddev=0.000070 bytes_write=845 bytes_read=5897 bytes_retransmit=9 bytes_invalid=0 s>
Stats 778.2: gcodein=67 mcu: mcu_awake=0.011 mcu_task_avg=0.000098 mcu_task_stddev=0.000070 bytes_write=851 bytes_read=6053 bytes_retransmit=9 bytes_invalid=0 s>
Stats 779.2: gcodein=67 mcu: mcu_awake=0.011 mcu_task_avg=0.000098 mcu_task_stddev=0.000070 bytes_write=857 bytes_read=6195 bytes_retransmit=9 bytes_invalid=0 s>
Stats 780.2: gcodein=67 mcu: mcu_awake=0.011 mcu_task_avg=0.000098 mcu_task_stddev=0.000070 bytes_write=863 bytes_read=6351 bytes_retransmit=9 bytes_invalid=0 s>
Requested temperature (-1.0) out of range (0.0:250.0)
Stats 781.2: gcodein=79 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=869 bytes_read=6535 bytes_retransmit=9 bytes_invalid=0 s>
Stats 782.2: gcodein=79 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=875 bytes_read=6677 bytes_retransmit=9 bytes_invalid=0 s>
Stats 783.2: gcodein=84 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=881 bytes_read=6833 bytes_retransmit=9 bytes_invalid=0 s>
Stats 784.2: gcodein=84 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=887 bytes_read=6991 bytes_retransmit=9 bytes_invalid=0 s>
Stats 785.2: gcodein=84 mcu: mcu_awake=0.010 mcu_task_avg=0.000097 mcu_task_stddev=0.000069 bytes_write=893 bytes_read=7123 bytes_retransmit=9 bytes_invalid=0 s>
Stats 786.2: gcodein=84 mcu: mcu_awake=0.010 mcu_task_avg=0.000095 mcu_task_stddev=0.000067 bytes_write=899 bytes_read=7282 bytes_retransmit=9 bytes_invalid=0 s>
Stats 787.2: gcodein=84 mcu: mcu_awake=0.010 mcu_task_avg=0.000095 mcu_task_stddev=0.000067 bytes_write=905 bytes_read=7440 bytes_retransmit=9 bytes_invalid=0 s>
Stats 788.2: gcodein=89 mcu: mcu_awake=0.010 mcu_task_avg=0.000095 mcu_task_stddev=0.000067 bytes_write=911 bytes_read=7569 bytes_retransmit=9 bytes_invalid=0 s>

mentaluproar, 2019-04-02:
Sorry, fixed the log.
[tehlog.txt](https://github.com/KevinOConnor/klipper/files/3032179/tehlog.txt)

ozligia, 2019-04-02:
To disable it, set it to zero, not minus one

mentaluproar, 2019-04-03:
Simplify3D doesn't work that way.  It looks like Cura doesn't even use the T argument, but instead uses T0/T1 before anything referencing the extruder.

ozligia, 2019-04-03:
Correct syntax for klipper is:
M109 [T<index>] S<temperature> (in case of single nozzle index not needed)
If simplify 3D generates other codes, try to look at firmware settings and select profile for marlin/sprinter/repetier

ozligia, 2019-04-03:
So, quick research.
Yes, Cura uses select tool command, that is also supported in klipper:
`T<index>`
and
`M109 [T<index>] S<temperature>`
So your only problem is that Simplify3D uses S-1
And that is not the problem in klipper.
Though you can try to set
`min_temp: -1`
in extruder section of config

KevinOConnor, 2019-04-06:
Odd that Simplify3d would generate such a non-standard request.  As noted, change your `min_temp` to a value less than or equal to -1 to avoid the error message.

-Kevin

mentaluproar, 2019-04-06:
I'm thinking it's a marlin thing, and it's just putting out code marlin understands.

KevinOConnor, 2019-04-21:
I'm going to close this issue as it looks like the discussion has concluded.

-Kevin

closed 2019-04-21 by KevinOConnor