Klipper3d/klipper #1486: Probe/BLTouch offset conventions

sgtnoodle, 2019-04-02:
I'm setting up a BLTouch for the first time and I'm struggling to wrap my head around the meaning of x_offset, y_offset and z_offset based on the documentation. I wonder if the convention is backwards from what folk expect on the x and y axes, and if that's the root cause of these other bed mesh related issues:

https://github.com/KevinOConnor/klipper/issues/1481
https://github.com/KevinOConnor/klipper/issues/1472

The top down view of my extruder is like this:

```
Y+
|
|        N
|             P
|
0------------------X+
```
Where N is "Nozzle" and P is "Probe"

The documentation for the BLTouch says to place the nozzle somewhere on the bed, record its position, move the probe to that point, take the difference.

Lets's say I do that and get N = (100, 100) and P = (90, 110). 100-90 = 10, 100 - 110 = -10. Therefore, the X offset should be 10 and the Y offset -10. Ok, that makes sense: the probe is to the right (+X) and down (-Y) from the nozzle. The x and y offset is effectively the position of the probe relative to the nozzle.

The front view of my extruder is like this (when the BLTouch is pin-down):

```
Z+
|
|        N
|             P
|
0------------------X+
```

The documentation says to start with a z_offset of 10, home, and then jog the Z axis down until it touches the plate. Subtract the final Z position from 10.

Ok, let's say I do that and I get the doc example of 8.2. 10 - 8.2 = 1.8. Therefore, the Z offset should be 1.8. This is where it get's weird. The probe is down (-Z) from the nozzle, but the Z offset is positive! Instead of the z offset being the position of the probe relative to the nozzle, it's the position of the nozzle relative to the probe.

I was helping my dad set up Klipper on his printer the other day, and his X and Y probe offsets from Marlin were seemingly of the convention "the nozzle's position relative to the probe". Based on Klipper's documentation, I had to flip the signs. If X and Y have an opposite convention from Z and from other 3D printer firmwares, perhaps it's a common mistake for folk to have their X and Y offsets wrong?

For what it's worth, here are my actual offsets:
```
[bltouch]
x_offset: 27.9
y_offset: -20.7
z_offset: 0.7
```

sgtnoodle, 2019-04-02:
Well, I looked it up in Marlin and it appears to follow the "Probe's offset from nozzle" convention in X, Y and Z. So, it's consistent with Klipper's X and Y but inconsistent with Z.

jakep82, 2019-04-02:
"Instead of the z offset being the position of the probe relative to the nozzle, it's the position of the nozzle relative to the probe."

The Z offset is neither of these. It's the position of the nozzle relative to the bed (aka Z0) when the probe triggers.

sgtnoodle, 2019-04-02:
By probe position I specifically mean the probe's trigger position. By definition, the probe's trigger position in Z is equal to the bed's position in Z when the probe triggers.

All I'm saying is that the conventions in the config file are confusing to me because the Z offset has a different meaning than the x and y offsets:
[probe] x_offset: The probe's x trigger position relative to the nozzle
[probe] y_offset: The probe's y trigger position relative to the nozzle
[probe] z_offset: The nozzle's Z position relative to the probe's z trigger position (said equivalently, the position of the nozzle relative to the bed when the probe triggers)

When there's per-axis settings of the similar names to define an offset in a 3D coordinate space, doesn't it make the most sense to use the same frame of reference for all 3 axes?

dmp450, 2019-04-02:
There's some inconsistencies in the documentation with how XY offsets are described. In the BLTouch section, it says to move the probe to where the nozzle is. In the probe section, it says to move the nozzle to where the probe is. These should be updated to say the same thing, and they should be described as you mentioned: the probe's position relative to the nozzle. "Take the difference" is a bit ambiguous. I got mixed instructions in the Discord group as well, and only found out it was the probe relative to the nozzle by looking through the source.

jakep82, 2019-04-02:
The instructions in the probe calibration document were written by @KevinOConnor.  They appear to be comprehensive and will yield the correct result.  The instructions in the BLTouch document were written by a contributor, and are less clear.  Simply copying the X/Y offset instructions from the probe document to the BLTouch document (and replacing the word probe with BLTouch) would solve this issue.

Arksine, 2019-04-03:
Its true that the z_offset doesn't use a distance relative to the nozzle, but rather an absolute distance between the nozzle and the bed.  I don't think it should be changed as it would break all current configurations, with results that could potentially be damaging.

I can see how making the documentation between the probe and bltouch sections might be a good idea.

dmp450, 2019-04-03:
I submitted a pull request to make the documentation between probe and bltouch match. I've never submitted a pull request before, so I hope I did it right. Looking at the documentation a second time though, I think it might be better to just put a link to the probe section in the bltouch documentation when it comes to finding offsets. The method isn't unique to the bltouch, so I'm not sure why it needs its own document. It also seems redundant to have it say the exact same thing as the probe section. Calculating the Z offset in the bltouch doc seems to a be bit more thorough than in the probe doc.

KevinOConnor, 2019-04-03:
@sgtnoodle - I can see where there might be some confusion given your description.  Not sure changing it makes sense though.

@dmp450 - I agree that a link may be better than duplicating the text.  As for the Z offset, in my opinion, the automated PROBE_CALIBRATE tool described in Probe_Calibrate.md (along with the "paper test" description in Bed_Level.md) is a better reference.  Both of these two documents were written after the BLTouch.md document and the BLTouch.md document hasn't been updated.

-Kevin

dmp450, 2019-04-03:
@KevinOConnor I'm no printing pro, but I found the advice in the bltouch document of printing a 50mm x 50mm single layer square and adjusting the offset as it prints to be helpful. Would it be beneficial to include that in Probe_Calibrate.md (or Bed_Level.md) in addition to adding a link to BLTouch.md?

sgtnoodle, 2019-04-05:
Having brewed on it a few days, here are my current thoughts.

A probe's x_offset and y_offset make the most sense to me as the probe's offset from the x and y axis origin. It just so happens that folk configure their axes (via position_endpoint parameters) so that the main extruder nozzle's xy offset is 0 (aka the nozzle's x y position is 0,0 after homing). You could imagine a configuration where the x and y axis origin is configured to be some other arbitrary reference point, say the probe. In that example, the probe's xy offset is 0 and the nozzle is offset from the probe using gcode commands (as is done for a 2nd extruder).

When a probe is used for z axis homing, the probe's z_offset is literally (in code) used to override the z axis's position_endpoint parameter. The `z_offset` parameter doesn't represent the probe's offset from the z axis origin, but rather the position of the z axis when the probe is triggered.

I get that what's done is done and no one wants to break folks' existing configurations. What do you think about deprecating the use of z_offset and instead using the [stepper_z] section's position_endstop? If a config omits a stepper_z position_endstop but has a probe z_offset, use z_offset as the position_endstop. If a config has a stepper_z position_endstop but omits a probe z_offset, use the position_endstop. If a config has both a stepper_z position_endstop and a probe z_offset, error out and require the config file be fixed.

KevinOConnor, 2021-03-05:
I'm going to close this.  Although I agree the current convention could be a little confusing, it doesn't look like something we'll likely change.

-Kevin

closed 2021-03-05 by KevinOConnor