Klipper3d/klipper #1487: Endstops triggered

dugly66, 2019-04-02:
OK had marlin running fine fore 2 years just trying to switch to klipper and see. Using the ramps example file Ive gotten screen working and the axis will move. Heaters working but the endstops are triggered open or not. is there anything that you can see in my log file?
[klippy.log](https://github.com/KevinOConnor/klipper/files/3034467/klippy.log)

AndyWo83, 2019-04-02:
Have you check the following document?
[klipper/docs/Config_checks.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md)

There is a section "Verify Endstops"

dugly66, 2019-04-02:
Yes i have. But then the question is do i have the correct pins assigned
because they never change states.

mjf55, 2019-04-02:
What board did you have configured in your MARLIN configuration.h file?  Can you then look that board up and see what pins Marlin used?
For giggles, my delta has simple switches as endstops and are configured as:
endstop_pin: ^!ar2 
endstop_pin: ^!ar15
endstop_pin: ^!ar19

dugly66, 2019-04-02:
Its a bog standard ramps 1.4 board so I figured the ramps config would be
correct.

dugly66, 2019-04-02:
OK here is a weird thing that makes no sense. I noticed the second pin comented out in the ramps config so I uncommented it and commented the other out. It then showed as being open. so I closed the enstop and it showed as being triggered (as expected) but then I opened it again and now its triggered again no matter what. Is my board possesed

mjf55, 2019-04-02:
Sounds like you need to enable the hardware pull up by putting the '^' in front of the pin name.

dugly66, 2019-04-03:
Ok Tried mjf55's suggestion same result. Redid the entire config and same result. I have no idea what I am doing wrong. Can someone with a ramps board send me there config to try?

AxMod3DPrint, 2019-04-03:
Can you pastebin your marlin config? We may be able to help further with
that.

dugly66, 2019-04-03:
https://pastebin.com/VjHBVmb8
that is the config that I just threw back on it and the endstops all work

AxMod3DPrint, 2019-04-03:
,,,
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
#endstop_pin: ^ar2
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
#endstop_pin: ^ar15
position_endstop: 0
position_max: 200
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^ar18
#endstop_pin: ^ar19
position_endstop: 0.5
position_max: 200
,,, 

The end stop pins as above should be correct. You just need to check if the switches are wired as normally open (NO) or normally closed (NC).  Off the top of my head I can't remember which needs to be inverted, but try one endstop only by removing or adding the exclamation mark (!) and holding it down and issuing a M119 at the same time in the console.

dugly66, 2019-04-05:
ok I reloaded klipper back onto it. now the x and y seem to be working only reversed. So from my understanding I would put an ! on the endstop_pin: line to reverse the behavior correct?

jakep82, 2019-04-05:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md#verify-endstops

dugly66, 2019-04-05:
Got it thank you all very very much. Now onto the probe for z. Hopefully I can figure that one out myself...

closed 2019-04-05 by dugly66