Klipper3d/klipper #1489: Probe triggered prior to movement

potchin, 2019-04-02:
I'm sure I'm missing something here. I'm trying to run `bed_mesh_calibrate` on my AM8 (fitted with a capacitive z sensor). The initial probe works fine but when I run the calibrate its not lifting away from the bed enough between probe points and gives a reasonable error of...

> Probe triggered prior to movement

I had this working at some point and have undoubtedly broken something tweaking the config. But i have no idea what.

Any suggestions?

```Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-386-g9bd3e80'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.9 (default, Sep 26 2018, 05:58:52) \n[GCC 4.9.2]'
Start printer at Tue Apr  2 19:39:06 2019 (1554233946.7 3193965.0)
===== Config file =====
[stepper_x]
step_pin = PD7
dir_pin = PC5
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC2
position_endstop = -30
position_max = 220
position_min = -30
homing_speed = 50

[stepper_y]
step_pin = PC6
dir_pin = PC7
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC3
position_endstop = -8
position_min = -8
position_max = 220
homing_speed = 50

[stepper_z]
step_pin = PB3
dir_pin = !PB2
enable_pin = !PA5
step_distance = .0025
endstop_pin = probe:z_virtual_endstop
position_min = -8
position_max = 240
homing_speed = 20

[probe]
pin = ^!PC4
x_offset = 65
y_offset = -36
z_offset = 4.4

[extruder]
step_pin = PB1
dir_pin = PB0
enable_pin = !PD6
step_distance = .0105
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PD5
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA7
min_temp = 0
max_temp = 250
pressure_advance = 0.125
max_extrude_cross_section = 0.8
control = pid
pid_kp = 28.278
pid_ki = 1.257
[... 123 lines cut ...]
Stats 3194001.3: gcodein=72 mcu: mcu_awake=0.023 mcu_task_avg=0.000145 mcu_task_stddev=0.000173 bytes_write=3385 bytes_read=7309 bytes_retransmit=9 bytes_invalid=0 send_seq=171 receive_seq=171 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000198 heater_bed: target=0 temp=13.5 pwm=0.000 print_time=1373.579 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.8 pwm=0.000
Stats 3194002.3: gcodein=72 mcu: mcu_awake=0.023 mcu_task_avg=0.000145 mcu_task_stddev=0.000173 bytes_write=3846 bytes_read=7486 bytes_retransmit=9 bytes_invalid=0 send_seq=182 receive_seq=182 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=52 stalled_bytes=0 freq=16000200 heater_bed: target=0 temp=13.6 pwm=0.000 print_time=1375.513 buffer_time=0.433 print_stall=0 extruder: target=0 temp=12.7 pwm=0.000
Stats 3194003.3: gcodein=72 mcu: mcu_awake=0.023 mcu_task_avg=0.000145 mcu_task_stddev=0.000173 bytes_write=3924 bytes_read=7629 bytes_retransmit=9 bytes_invalid=0 send_seq=186 receive_seq=185 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000198 heater_bed: target=0 temp=13.5 pwm=0.000 print_time=1375.513 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.9 pwm=0.000
Stats 3194004.3: gcodein=82 mcu: mcu_awake=0.023 mcu_task_avg=0.000145 mcu_task_stddev=0.000173 bytes_write=4234 bytes_read=7778 bytes_retransmit=9 bytes_invalid=0 send_seq=193 receive_seq=193 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000203 heater_bed: target=0 temp=13.6 pwm=0.000 print_time=1377.631 buffer_time=0.548 print_stall=0 extruder: target=0 temp=12.8 pwm=0.000
Stats 3194005.3: gcodein=82 mcu: mcu_awake=0.023 mcu_task_avg=0.000145 mcu_task_stddev=0.000173 bytes_write=4240 bytes_read=7878 bytes_retransmit=9 bytes_invalid=0 send_seq=194 receive_seq=194 retransmit_seq=2 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000203 heater_bed: target=0 temp=13.6 pwm=0.000 print_time=1377.631 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.8 pwm=0.000
Stats 3194006.3: gcodein=101 mcu: mcu_awake=0.016 mcu_task_avg=0.000128 mcu_task_stddev=0.000151 bytes_write=4993 bytes_read=8072 bytes_retransmit=9 bytes_invalid=0 send_seq=208 receive_seq=208 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000207 heater_bed: target=0 temp=13.9 pwm=0.000 print_time=1381.381 buffer_time=2.297 print_stall=0 extruder: target=0 temp=12.7 pwm=0.000
Stats 3194007.3: gcodein=101 mcu: mcu_awake=0.016 mcu_task_avg=0.000128 mcu_task_stddev=0.000151 bytes_write=4999 bytes_read=8186 bytes_retransmit=9 bytes_invalid=0 send_seq=209 receive_seq=209 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000204 heater_bed: target=0 temp=13.7 pwm=0.000 print_time=1381.381 buffer_time=1.296 print_stall=0 extruder: target=0 temp=12.8 pwm=0.000
Stats 3194008.3: gcodein=101 mcu: mcu_awake=0.016 mcu_task_avg=0.000128 mcu_task_stddev=0.000151 bytes_write=5005 bytes_read=8286 bytes_retransmit=9 bytes_invalid=0 send_seq=210 receive_seq=210 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000205 heater_bed: target=0 temp=13.9 pwm=0.000 print_time=1381.381 buffer_time=0.296 print_stall=0 extruder: target=0 temp=12.8 pwm=0.000
Probe triggered prior to movement
Probe triggered prior to movement
Stats 3194009.3: gcodein=106 mcu: mcu_awake=0.016 mcu_task_avg=0.000128 mcu_task_stddev=0.000151 bytes_write=5057 bytes_read=8439 bytes_retransmit=9 bytes_invalid=0 send_seq=215 receive_seq=215 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000205 heater_bed: target=0 temp=13.7 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.9 pwm=0.000
Stats 3194010.3: gcodein=106 mcu: mcu_awake=0.016 mcu_task_avg=0.000128 mcu_task_stddev=0.000151 bytes_write=5095 bytes_read=8563 bytes_retransmit=9 bytes_invalid=0 send_seq=218 receive_seq=218 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000204 heater_bed: target=0 temp=13.5 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.9 pwm=0.000
Stats 3194011.3: gcodein=106 mcu: mcu_awake=0.016 mcu_task_avg=0.000130 mcu_task_stddev=0.000165 bytes_write=5101 bytes_read=8678 bytes_retransmit=9 bytes_invalid=0 send_seq=219 receive_seq=219 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000203 heater_bed: target=0 temp=13.6 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.9 pwm=0.000
Stats 3194012.3: gcodein=106 mcu: mcu_awake=0.016 mcu_task_avg=0.000130 mcu_task_stddev=0.000165 bytes_write=5107 bytes_read=8792 bytes_retransmit=9 bytes_invalid=0 send_seq=220 receive_seq=220 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000202 heater_bed: target=0 temp=13.9 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=13.1 pwm=0.000
Stats 3194013.3: gcodein=106 mcu: mcu_awake=0.016 mcu_task_avg=0.000130 mcu_task_stddev=0.000165 bytes_write=5113 bytes_read=8906 bytes_retransmit=9 bytes_invalid=0 send_seq=221 receive_seq=221 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000202 heater_bed: target=0 temp=13.5 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.8 pwm=0.000
Stats 3194014.3: gcodein=111 mcu: mcu_awake=0.016 mcu_task_avg=0.000130 mcu_task_stddev=0.000165 bytes_write=5119 bytes_read=9006 bytes_retransmit=9 bytes_invalid=0 send_seq=222 receive_seq=222 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000202 heater_bed: target=0 temp=13.8 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.8 pwm=0.000
Stats 3194015.3: gcodein=111 mcu: mcu_awake=0.016 mcu_task_avg=0.000130 mcu_task_stddev=0.000165 bytes_write=5125 bytes_read=9120 bytes_retransmit=9 bytes_invalid=0 send_seq=223 receive_seq=223 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000202 heater_bed: target=0 temp=13.7 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.6 pwm=0.000
Stats 3194016.3: gcodein=111 mcu: mcu_awake=0.007 mcu_task_avg=0.000076 mcu_task_stddev=0.000065 bytes_write=5131 bytes_read=9247 bytes_retransmit=9 bytes_invalid=0 send_seq=224 receive_seq=224 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000201 heater_bed: target=0 temp=13.6 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_st [... cut ...] 381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.9 pwm=0.000
Stats 3194021.3: gcodein=116 mcu: mcu_awake=0.007 mcu_task_avg=0.000075 mcu_task_stddev=0.000062 bytes_write=5161 bytes_read=9802 bytes_retransmit=9 bytes_invalid=0 send_seq=229 receive_seq=229 retransmit_seq=2 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000198 heater_bed: target=0 temp=13.9 pwm=0.000 print_time=1381.381 buffer_time=0.000 print_stall=0 extruder: target=0 temp=12.9 pwm=0.000

```

jakep82, 2019-04-02:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L147

potchin, 2019-04-02:
That'll be it. Thanks

closed 2019-04-02 by potchin

pr3sidentspence, 2019-07-31:
@jakep82 The line linked here doesn't seem to be relevant (anymore?). What was the solution?

jakep82, 2019-07-31:
https://github.com/KevinOConnor/klipper/blob/2e16be224efcff315dafaf1f4e4b85e3c7a8c3c8/config/example-extras.cfg#L147