Klipper3d/klipper #1491: homing_override doesn't always use absolute positioning

sgtnoodle, 2019-04-03:
I recently started using [homing_override]:
```
[homing_override]
axes: z
gcode:
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X-6 Y46 F6000
    G28 Z
set_position_z: 0
```
I noticed some funny behavior where the G1 X-6 Y46 would sometimes result in an out-of-bounds error. I changed it to X0 Y46, and the axes would move but sometimes they would move to two different places (one of them causing a head crash!).

It turns out that, when pressing the "home Z" button in octoprint, it sends a G91 before the G28 Z, aka relative positioning mode. The gcode section in my homing_override would do a relative move rather than an absolute one.

I added G90 to my homing_override gcode section and all is well now. None of the [homing_override] examples I've seen in the documentation mention this, though, so I'll bet lots of people have buggy homing_override sections. Perhaps it would be better if the homing_override gcode section always ran in a predictable, clean context regardless of what gcodes were sent before it. Alternatively, maybe the example-extras.cfg and bltouch doc should be updated to mention the importance of issuing a G90 in the homing routine.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3036813/klippy.log)

Arksine, 2019-04-03:
If Octoprint is sending a G91 then that seems like more of an issue with Octoprint than Klipper.

KevinOConnor, 2019-04-03:
FYI, we do document the G90 in config/sample-probe-as-z-endstop.cfg .

FWIW, gcode is absolutely horrible - and G90/G91 is a glaring example.  I find it difficult to document how awful g-code is though.  (It's basically a never ending topic.)  If you have some specific changes that you think would help the documentation then let me know.

-Kevin

sgtnoodle, 2019-04-03:
I agree that it seems kind of silly of octoprint to do that, but I don't think it's invalid behavior.

It seems to me that the expected result of a G28 command is properly homed axes. It shouldn't matter whether the gcode parser is in an absolute mode or relative mode when the G28 command is issued.

A vanilla G28 implementation in klipper appears to be declarative in nature; the hard-coded homing routine uses the endpoint parameters for the axes in the config file and "does the right thing", likely without mutating the gcode parser's mode.

sgtnoodle, 2019-04-03:
I'll make a PR later today with some documentation fixes.

If you wanted to make the homing override more robust in code, it would be great if the homing override routine saved the current positioning mode, forced the parser into absolute mode before running the gcode script, and then restored the original mode at the end. I might take a crack at that myself, but will take me longer. :-)

KevinOConnor, 2019-04-03:
Adding SAVE_GCODE_STATE and RESTORE_GCODE_STATE commands would be useful.  We could then recommend that users deploy those commands in the various gcode macros.  @wizhippo had some code for something similar in PR #222.

-Kevin

sgtnoodle, 2019-04-04:
Here's a PR for some minor documentation improvements that would have helped in my case. I haven't contributed before, so hopefully I formatted it right!

https://github.com/KevinOConnor/klipper/pull/1495

KevinOConnor, 2019-06-07:
Hopefully  this should be addressed now.  (There's been a bunch of commits  recently adding SAVE/RESTORE_GCODE_STATE, and various documentation  updates.)

 -Kevin

closed 2019-06-18 by KevinOConnor