Klipper3d/klipper #1496: Heater extruder not heating at expected rate

noxx2015, 2019-04-04:
Hello, 

cant heat my extruder. get this 
Heater extruder not heating at expected rate

[klippy.log](https://github.com/KevinOConnor/klipper/files/3044623/klippy.log)

jakep82, 2019-04-04:
Your extruder isn't heating. Either your heater cartridge is bad, you have a wiring issue, your mosfet is bad, or you've configured the wrong heater_pin.

KevinOConnor, 2019-04-14:
I'm going to close this as I think the original question was answered (the code is correctly reporting that the extruder is not heating - likely an error in the config).  See https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md for some steps to help diagnose.

-Kevin

closed 2019-04-14 by KevinOConnor