Klipper3d/klipper #1497: Feature request - better kill switch support

Sidic101, 2019-04-04:
Right now, Klipper uses the M112 to put the printer in a "emergency shutdown" state. While that is great, this command cannot be used to, say, kill a printer move during printing when you accidentally select the wrong gcode. 

Using the "kill" button  (for example on the Reprapdiscount 2004 display) to reset the MCU does not work either, because the the fans switches off so filament can melt inside the heatbreak, causing a clog.

Inverting the fan behaviour in the MCU firmware, so that the fan is on after a restart and only switches off after the firmware connects to the Pi, could make the reset/kill switch work much better, because now you don't need to worry about heat creep when you kill a printer move in an emergency.

jakep82, 2019-04-04:
Klipper responds to M112 immediately. I use it all the time when I start a print and realize I've made a mistake during slicing.

Sidic101, 2019-04-04:
Than I must be doing something wrong, because it does not do that for me. After I start a print, it can only be stopped with "cancel print" and that takes a while as the pri ter first finishes the current move.

jakep82, 2019-04-04:
How are you sending the M112?  I generally do it from the terminal in Octoprint.

Sidic101, 2019-04-04:
The terminal stops working when a file is printing.

jakep82, 2019-04-04:
It should continue working.  You can also install the OctoKlipper plugin which will add a kill button to Octoprint which will execute M112.

Sidic101, 2019-04-04:
I have tried the plugin, it does not work for me.

jakep82, 2019-04-04:
I would guess you have configuration issue in Octoprint, but I'm not at my printer to verify.  The terminal should still respond during prints.

jakep82, 2019-04-04:
It appears this is a bug in Octoprint.  I forgot that a previous issue exists.  I don't have this problem because I use virtual SD for printing which is not affected by this bug.

https://github.com/KevinOConnor/klipper/issues/1081

Sidic101, 2019-04-05:
Either way, having the physical kill button work better would certainly be beneficial.

KevinOConnor, 2019-04-08:
I'm not sure what is being requested here.  If Klipper does receive an M112, it halts immediately - there is a known bug with OctoPrint however - see #1081.  If a "kill button" is configured in a Klipper display then it will internally issue an M112 and halt immediately - however some boards are known to directly map the kill button to an internal micro-controller reset.  On those boards, there's nothing Klipper can do to catch the kill button.  Finally, if you wish to enable a pin at micro-controller startup then that is possible via "make menuconfig" - enable the low-level options to see the relevant config setting.

-Kevin

Sidic101, 2019-04-16:
Sorry for the confusion (and for not replying earlier). 
What is being requested: make it so the fan that cools the cold end is on when the MCU program starts and only stops spinning after a connection to Klipper on the Pi is established. Doing so would mean that on the boards that have the kill button wired to the reset pin, the kill button would effectively do the exact same thing the M112 command does, which means that the switch on such boards can be used without altering the hardware.

KevinOConnor, 2019-04-21:
As noted above, it is possible to configure that via "make menuconfig".

-Kevin

Sidic101, 2019-04-25:
Ok, thanks. It§s not exactly what I had in mind but it should work well enough.

closed 2019-04-25 by Sidic101