Klipper3d/klipper #1498: SKR V1.3: I can't translate the display pins from Marlin to Klipper.

mtsgsd, 2019-04-04:
there are no errors in the log file.
I know there are people successfully hooking up a stock Creality display to EXP1 on the new  SKR V1.3 board using Marlin, but I can not work out how to map this in Klipper using its' pin names.
This is what I've attempted to do. but it's not working. I do get the backlight, but that's it. Any help would be greatly appreciated. Those of us on the Klipper Discord haven't figured this out yet.

BigTreeTech describes the EXP1 connector as follows:
                   5v      GND
(lcd_D7)     1.23    1.22 (LCD_D6)
(lcd_D5)     1.21    1.20 (LCD__D4)
(lcd_RS)     1.19    1.18 (LCD_EN
(BTN_ENC) 0.28    1.30 (BEEPER)

The config I used for Klipper for my MKS Gen L board was as follows:
# 128x64 Full Graphic Creality CR10 / ENDER 3 stockdisplay
#[display]
#lcd_type: st7920
#cs_pin: ar27
#sclk_pin: ar25
#sid_pin: ar29
#encoder_pins: ^ar23, ^ar17
#click_pin: ^!ar35

Marlin's pin def for the SKR board is the following:
#if ENABLED(CR10_STOCKDISPLAY)
    #define LCD_PINS_RS     P1_22
    #define BTN_EN1         P1_18
    #define BTN_EN2         P1_20
    #define LCD_PINS_ENABLE P1_23
    #define LCD_PINS_D4     P1_21

Translating pin for pin from the Ramps board I was using to the SKR board I get the following:
generic ramps:
      #if ENABLED(CR10_STOCKDISPLAY)
        #define BTN_EN1         17  >>1.18
        #define BTN_EN2         23 >>1.20
      #define LCD_PINS_RS       27 >>1.22
      #define LCD_PINS_ENABLE   29 >>1.23
      #define LCD_PINS_D4       25 >>1.21

      #define BTN_ENC           35 >> 0.28
      #define SD_DETECT_PIN     49 >> na
      #define KILL_PIN          41 >> na

mattthebaker, 2019-04-04:
Github issues are for bugs...

Try this:

```
[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.20
sid_pin: P1.18
encoder_pins: ^P3.25, ^P3.26
click_pin: ^!P0.28
```

Invert the encoder/click pins if they don't work right.

mattthebaker, 2019-04-04:
Github issues are for bugs.

Looks like you want this:
[display]
lcd_type: st7920
cs_pin: P1.22
sclk_pin: P1.21
sid_pin: P1.23
encoder_pins: ^P1.18, ^P1.20
click_pin: ^!P0.28

jakep82, 2019-04-04:
The CR-10 display does not have the same pinouts on the connectors based on what I've seen other places.  Assuming the Marlin pin definition you provided is correct, it would actually use the pins below.  I don't have this hardware so I can't confirm.

```
[display]
lcd_type: st7920
cs_pin: P1.22
sclk_pin: P1.21
sid_pin: P1.23
encoder_pins: ^P1.18, ^P1.20
click_pin: ^!P0.28
```

mtsgsd, 2019-04-04:
I should not have included that log as it shows a test I was doing. I started with the same layout you are both recommending. I'll just do without.

dreamcryer, 2019-04-05:
I have my SKR v1.3 but have not had time to play with it. Did you try this?
```[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.20
sid_pin: P1.18
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P0.28
```
from pull request <https://github.com/KevinOConnor/klipper/pull/1344>

mattthebaker, 2019-04-05:
cs, sclk, and sid should be all the pins necessary for serial communication, and we know which three marlin uses P1.21, P1.22, and P1.23. That is six possible combinations. If none work something is off about the marlin config.

KevinOConnor, 2019-04-21:
I'm going to close this as it looks like the discussion has concluded.  (To the best of my knowledge, Klipper works as intended and other users are reporting success with displays on the skr.)

-Kevin

closed 2019-04-21 by KevinOConnor

mizio85, 2019-05-24:
I have the same problem on display uc1701 (creality CR-20) I can't find the right pins, can anyone help me?

stevenvo, 2020-02-16:
> The CR-10 display does not have the same pinouts on the connectors based on what I've seen other places. Assuming the Marlin pin definition you provided is correct, it would actually use the pins [...]

this works for me